micropython/micropython #3476: MINGW64 make deploy could not claim interface 0

jpliew, 2017-12-08:
```make BOARD=STM32F4DISC deploy
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Writing build-STM32F4DISC/firmware.dfu to the board
Traceback (most recent call last):
  File "../../tools/pydfu.py", line 540, in <module>
    main()
  File "../../tools/pydfu.py", line 520, in main
    init()
  File "../../tools/pydfu.py", line 86, in init
    usb.util.claim_interface(__dev, __DFU_INTERFACE)
  File "build\bdist.win32\egg\usb\util.py", line 205, in claim_interface
  File "build\bdist.win32\egg\usb\core.py", line 102, in wrapper
  File "build\bdist.win32\egg\usb\core.py", line 167, in managed_claim_interface

  File "build\bdist.win32\egg\usb\backend\libusb0.py", line 521, in claim_interface
  File "build\bdist.win32\egg\usb\backend\libusb0.py", line 431, in _check usb.core.USBError: [Errno None] usb_claim_interface: could not claim interface 0, invalid configuration 0
make: *** [deploy] Error 1
```
adding `__dev.set_configuration()` at pydfu.py fixes the problem

```
    if len(devices) > 1:
        raise ValueError("Multiple DFU devices found")
    __dev = devices[0]
    __dev.set_configuration()
```
Successful result below

```
make BOARD=STM32F4DISC deploy
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosiy.
Writing build-STM32F4DISC/firmware.dfu to the board
File: build-STM32F4DISC/firmware.dfu
    DfuSe v1, image size: 312005, targets: 1
    Target 0, alt setting: 0, name: "ST...", size: 311720, elements: 2
      0, address: 0x08000000, size: 14952
      1, address: 0x08020000, size: 296752
    usb: 0483:df11, device: 0x0000, dfu: 0x011a, UFD, 16, 0x3aba588c
Writing memory...
0x08000000   14952 [=========================] 100%
0x08020000  296752 [=========================] 100%
Exiting DFU...
```

dpgeorge, 2017-12-13:
Thanks for the report and fix.  Do you know why/how this fixes the issue?  In particular, would adding that `set_configuration()` line have an adverse effect on other systems that do currently work?

jpliew, 2017-12-13:
This is a common issue with packages that is using PyUSB, I believed is an inherited issue since a long time. Googling this leads to lots of results https://www.google.com.au/search?q=usb_claim_interface%3A+could+not+claim+interface+0&oq=usb_claim_interface%3A+could+not+claim+interface+0

From PyUSB's tutorial, https://github.com/pyusb/pyusb/blob/master/docs/tutorial.rst

>  PyUSB tries to make your life as easy as possible. For example, after getting your device object, one of the first things you need to do before communicating with it is issuing a **_set_configuration_** request.

I am not sure why other OS works without this line, but from PyUSB's documentation, this line is required before communication.

dpgeorge, 2017-12-13:
Thanks for the details.  I checked on Linux and pydfu.py still works with this line so it's been added in bb516af1eb97d0c24fa12b00bfde196b3db94c66

closed 2017-12-13 by dpgeorge