micropython/micropython #3478: persistentcode.c: save_raw_code() - add finalizer

adritium, 2017-12-08:
https://github.com/micropython/micropython/blob/e9d29c9ba997ae0f00f16f3a21dffce7c763a3d4/py/persistentcode.c#L319-L371

`save_raw_code` needs a finalizing call to `mp_print_t *print` for the case of `print` being a function that saves to flash (or more general, needs a manual flush).

Flash can only be written one page at a time so unless you just happen to call `print` with a whole number page multiple of bytes . . .  `print` will be stuck waiting for the last (**page_size - buffer_curr_num_bytes**) bytes forever.

With a filesystem, this is not an issue because the OS will flush the buffer at some point but even in that case, it certainly wouldn't hurt to have a finalizing call.

The simplest way to implement this flush is to call `print->print_strn` with `NULL` for data and `0` for length (unless someone can think of valid use for those set of arguments).

`print->print_strn(print->data, (const char*)NULL, 0);`

But the cleaner way would be to add a member to `mp_print_t`

https://github.com/micropython/micropython/blob/e9d29c9ba997ae0f00f16f3a21dffce7c763a3d4/py/mpprint.h#L48-L53

```
typedef void (*mp_print_strn_t)(void *data, const char *str, size_t len);
typedef void (*mp_print_final_t)(void *data);

typedef struct _mp_print_t {
    void *data;
    mp_print_final_t print_final;
    mp_print_strn_t print_strn;
} mp_print_t;
```

And the very last statement in `mp_save_raw_code` is `print->print_final(print->data)`

dpgeorge, 2017-12-13:
The function that calls `save_raw_code()` (actually that calls `mp_raw_code_save()`) should be responsible for doing any finalisation of the stream.  For example, `mp_raw_code_save_file()` closes (and implicitly flushes) the file once done.

adritium, 2017-12-20:
> For example, mp_raw_code_save_file() closes (and implicitly flushes) the file once done.

I think you made **my** point with this example: I agree that it makes sense for `mp_raw_code_save_file()` to call `close(fd)`. For that same reason, `mp_raw_code_save()` should also call the finalizer: either the finalizer is called by the caller of `mp_raw_code_save*()` or it's called inside said function.

dpgeorge, 2017-12-22:
> I agree that it makes sense for mp_raw_code_save_file() to call close(fd). For that same reason, mp_raw_code_save() should also call the finalizer:

I don't see how this logically follows.  `mp_raw_code_save()` is essentially a helper function and the caller of it should finalise the stream, because the caller knows best how to do finalisation.

adritium, 2017-12-26:
`mp_raw_code_save_file()` calls the finalizer `close()` so why shouldn’t `mp_raw_code_save()` call a finalizer?

In the same way that `mp_print_t` provides the interface to “print” which needs to be set to an actual “print” function, `mp_print_t` should also contain an interface to finalize (as I mentioned in the first post to this issue)

dpgeorge, 2017-12-28:
> mp_raw_code_save_file() calls the finalizer close() so why shouldn’t mp_raw_code_save() call a finalizer?

Becasue `mp_raw_code_save_file()` can do it and there's no need to burden the core routine with a finaliser call.

In your specific case, it looks like it should work just fine to flush the output at the very end (ie not in `mp_raw_code_save()`).

dpgeorge, 2021-05-18:
`mp_raw_code_save()` is a mid-level API function (eg you may want to call it twice in a row to save 2 separate raw codes).  Higher-level API functions built on top of this function are responsible for flushing/closing the stream, eg:
```c
void mp_raw_code_save_flash(...) {
    mp_raw_code_save(...);
    commit_to_flash();
}
```

closed 2021-05-18 by dpgeorge