micropython/micropython #3480: Add function to merge two mp_raw_code_t

adritium, 2017-12-08:
If you compile two blocks separately, you'll get two separate `mp_raw_code_t`. 
It should be possible to merge them into one `mp_raw_code_t`.

dpgeorge, 2017-12-13:
> It should be possible to merge them into one mp_raw_code_t.

That doesn't really make sense.  An `mp_raw_code_t` represents one Python scope (like a function, or class body) and these can't be merged.

adritium, 2017-12-19:
You have a Python file that cannot be compiled on the module because it's too big. However, you can use the fact that compilation of atomic blocks are independent.

**main.py**
```
import pyb

led = pyb.LED(1)

while True:
	led.toggle()

```

**frozen_mpy_main.c**
```
STATIC const byte bytecode_data_main__lt_module_gt_[51] = {
    0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0a,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_main_dot_py & 0xff, MP_QSTR_main_dot_py >> 8,
    0x49, 0x4c, 0x20, 0x00, 0x00, 0xff,
    0x80, 
    0x11, 
    0x68, MP_QSTR_pyb & 0xff, MP_QSTR_pyb >> 8,
    0x24, MP_QSTR_pyb & 0xff, MP_QSTR_pyb >> 8,
    0x1b, MP_QSTR_pyb & 0xff, MP_QSTR_pyb >> 8,
    0x1e, MP_QSTR_LED & 0xff, MP_QSTR_LED >> 8,
    0x81, 
    0x66, 0x01, 
    0x24, MP_QSTR_led & 0xff, MP_QSTR_led >> 8,
    0x1b, MP_QSTR_led & 0xff, MP_QSTR_led >> 8,
    0x1e, MP_QSTR_toggle & 0xff, MP_QSTR_toggle >> 8,
    0x66, 0x00, 
    0x32, 
    0x35, 0xf4, 0x7f, 
    0x11, 
    0x5b, 
};
```

**main1.py**
```
import pyb
led = pyb.LED(1)
```

**frozen_mpy_main1.c**
```
// frozen bytecode for file main1.py, scope main1_<module>
STATIC const byte bytecode_data_main1__lt_module_gt_[37] = {
    0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x08,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_main1_dot_py & 0xff, MP_QSTR_main1_dot_py >> 8,
    0x49, 0x00, 0x00, 0xff,
    0x80, 
    0x11, 
    0x68, MP_QSTR_pyb & 0xff, MP_QSTR_pyb >> 8,
    0x24, MP_QSTR_pyb & 0xff, MP_QSTR_pyb >> 8,
    0x1b, MP_QSTR_pyb & 0xff, MP_QSTR_pyb >> 8,
    0x1e, MP_QSTR_LED & 0xff, MP_QSTR_LED >> 8,
    0x81, 
    0x66, 0x01, 
    0x24, MP_QSTR_led & 0xff, MP_QSTR_led >> 8,
    0x11, 
    0x5b, 
};
```

**main2.py**
```
while True:
	led.toggle()
```

**frozen_mpy_main2.c**
```
// frozen bytecode for file main2.py, scope main2_<module>
STATIC const byte bytecode_data_main2__lt_module_gt_[29] = {
    0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x08,
    MP_QSTR__lt_module_gt_ & 0xff, MP_QSTR__lt_module_gt_ >> 8,
    MP_QSTR_main2_dot_py & 0xff, MP_QSTR_main2_dot_py >> 8,
    0x21, 0x00, 0x00, 0xff,
    0x1b, MP_QSTR_led & 0xff, MP_QSTR_led >> 8,
    0x1e, MP_QSTR_toggle & 0xff, MP_QSTR_toggle >> 8,
    0x66, 0x00, 
    0x32, 
    0x35, 0xf4, 0x7f, 
    0x11, 
    0x5b, 
};
```

dpgeorge, 2017-12-22:
It's not easy to merge blocks of compiled code because the prelude needs to change.  In particular the bytecode-to-line-number mapping must be merged and that's not a simple case of just concatenating the data.

dpgeorge, 2017-12-22:
@adritium it seems you have edited this issue to the point where it's really different from the original.  For example the quote in the second post no longer exists in the first post and it's confusing to read.

This issue now seems to be about reducing the RAM using of the parser.  Please see #2417 for ideas on that.

closed 2018-05-02 by adritium