micropython/micropython #3481: Unable to get working firmware since GCC 6.4

kamikaze, 2017-12-09:
Firmware compiles ok, but PyBoard shows blue, yellow and red leds on and nothing works. Official firmwares work fine. [There](https://forum.micropython.org/viewtopic.php?p=23729#p23729) are people that are able to build with 7.2, so I'm just curious what is wrong in my case.

I'm using Gentoo Linux and have installed toolchain with:
`crossdev --target arm-none-eabi -s4 --genv "EXTRA_ECONF=--disable-libstdc++-v3"`

Then I'm doing following:
`
$ LC_ALL=C git clone https://github.com/micropython/micropython.git
Cloning into 'micropython'...
remote: Counting objects: 58587, done.
remote: Compressing objects: 100% (49/49), done.
remote: Total 58587 (delta 54), reused 77 (delta 53), pack-reused 58485
Receiving objects: 100% (58587/58587), 32.12 MiB | 5.43 MiB/s, done.
Resolving deltas: 100% (42691/42691), done.
`

`
$ cd micropython
$ LC_ALL=C git submodule update --init
Submodule 'lib/axtls' (https://github.com/pfalcon/axtls) registered for path 'lib/axtls'
Submodule 'lib/berkeley-db-1.xx' (https://github.com/pfalcon/berkeley-db-1.xx) registered for path 'lib/berkeley-db-1.xx'
Submodule 'lib/libffi' (https://github.com/atgreen/libffi) registered for path 'lib/libffi'
Submodule 'lib/lwip' (http://git.savannah.gnu.org/r/lwip.git) registered for path 'lib/lwip'
Submodule 'lib/stm32lib' (https://github.com/micropython/stm32lib) registered for path 'lib/stm32lib'
Cloning into '/run/media/kamikaze/e7b9379e-9c7a-497d-8c39-ece0683cdb30/projects/micropython/lib/axtls'...
Cloning into '/run/media/kamikaze/e7b9379e-9c7a-497d-8c39-ece0683cdb30/projects/micropython/lib/berkeley-db-1.xx'...
Cloning into '/run/media/kamikaze/e7b9379e-9c7a-497d-8c39-ece0683cdb30/projects/micropython/lib/libffi'...
Cloning into '/run/media/kamikaze/e7b9379e-9c7a-497d-8c39-ece0683cdb30/projects/micropython/lib/lwip'...
Cloning into '/run/media/kamikaze/e7b9379e-9c7a-497d-8c39-ece0683cdb30/projects/micropython/lib/stm32lib'...
Submodule path 'lib/axtls': checked out '43a6e6bd3bbc03dc501e16b89fba0ef042ed3ea0'
Submodule path 'lib/berkeley-db-1.xx': checked out '35aaec4418ad78628a3b935885dd189d41ce779b'
Submodule path 'lib/libffi': checked out 'e9de7e35f2339598b16cbb375f9992643ed81209'
Submodule path 'lib/lwip': checked out '5b8b5d459e7dd890724515bbfad86c705234f9ec'
Submodule path 'lib/stm32lib': checked out 'd2bcfda543d3b99361e44112aca929225bdcc07f'
`

`
$ cd ports/stm32/
$ LC_ALL=C make BOARD=PYBV11 V=1 deploy
`
Output is here: https://gist.github.com/kamikaze/3cc0b8098ae0f4df18249716be087fe6

GCC version:
`
$ LC_ALL=C arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/arm-none-eabi/7.2.0/lto-wrapper
Target: arm-none-eabi
Configured with: /var/tmp/portage/cross-arm-none-eabi/gcc-7.2.0/work/gcc-7.2.0/configure --host=x86_64-pc-linux-gnu --target=arm-none-eabi --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/arm-none-eabi/gcc-bin/7.2.0 --includedir=/usr/lib/gcc/arm-none-eabi/7.2.0/include --datadir=/usr/share/gcc-data/arm-none-eabi/7.2.0 --mandir=/usr/share/gcc-data/arm-none-eabi/7.2.0/man --infodir=/usr/share/gcc-data/arm-none-eabi/7.2.0/info --with-gxx-include-dir=/usr/lib/gcc/arm-none-eabi/7.2.0/include/g++-v7 --with-python-dir=/share/gcc-data/arm-none-eabi/7.2.0/python --enable-languages=c,c++ --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --enable-nls --without-included-gettext --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo 7.2.0 p1.1' --disable-esp --enable-libstdcxx-time --enable-poison-system-directories --with-sysroot=/usr/arm-none-eabi --disable-bootstrap --with-newlib --enable-multilib --disable-altivec --disable-fixed-point --disable-libgcj --disable-libgomp --disable-libmudflap --disable-libssp --disable-libcilkrts --disable-libmpx --disable-vtable-verify --disable-libvtv --disable-libquadmath --enable-lto --without-isl --disable-libsanitizer --enable-default-pie --enable-default-ssp --disable-libstdc++-v3
Thread model: single
gcc version 7.2.0 (Gentoo 7.2.0 p1.1)
`

adritium, 2017-12-10:
Use `objdump` or [`readelf`](https://sourceware.org/binutils/docs/binutils/readelf.html) on the `.elf`s from working and non-working toolchain and compare the outputs.

adritium, 2017-12-12:
I’m not sure if you added that gist link but I was talking about doing a diff between objdump dumps of one working elf and one non-working.

dpgeorge, 2017-12-14:
The thing that stands out in the log of the build is the size of the firmware:
```
   text	   data	    bss	    dec	    hex	filename
 299412	  42328	  28268	 370008	  5a558	build-PYBV11/firmware.elf
```
The data should be around 36 bytes and the text around 323000 bytes.  This discrepancy points to using PIC/PIE.  And indeed your arm-none-eabi-gcc is compiled with ` --enable-default-pie`.  This needs to be explicitly disable, something like `-fno-pie`.

kamikaze, 2017-12-15:
Thanks, that worked: `$ LC_ALL=C make BOARD=PYBV11 V=1 CFLAGS_EXTRA="-fno-pie"`

closed 2017-12-15 by kamikaze