micropython/micropython #3482: Support ssd1306 with hardware I2C on stm32 port

jimmo, 2017-12-10:
I'd like to use this OLED driver: https://github.com/micropython/micropython/blob/master/drivers/display/ssd1306.py on a pyboard with I2C.

However, this driver expects to be using the soft I2C which provides the "primitive" start/write/stop API (and it does in fact work great on my pyboard). It makes sense that it's using that API - it wants to transmit a single byte followed by a framebuffer. I can't see a way to do this using the writeto() API that doesn't involve a big copy.

I use this for teaching school students so I'm trying to avoid mysterious traps like understanding the difference between `machine.I2C('X')` vs `machine.I2C(scl=...,sda=...)`.

I have a few suggestions:

* Add a scatter/gather version of writeto (i.e. the buf argument is allowed to be an iterable of buffers) and update ssd1306.py to use it.

* Have the stm32 hardware I2C implementation fall back to software I2C for start/stop/readinto/write. (Not sure this is a good idea...).

* Add these methods to the stm32 hardware implementation. POC for start/stop/write at https://github.com/micropython/micropython/compare/master...jimmo:pyb-i2c-ll?expand=1  (This works for my display). It is a bit complicated because of the address handling though. It would be simpler if `start()` took the 7-bit address and then figured out what do based on whether `read` or `write` was called. (What is the policy on making code in micropython/drivers depend on newer firmware versions?).

Thanks!

peterhinch, 2017-12-10:
I'm puzzled by the title of this issue. The `SSD1306_I2C` constructor requires an I2C instance argument. This may be a hardware or software object depending on how the user code instantiated it. 

The changes you propose are to `i2c.c` and seem to have little to do with `ssd1306.py`. Perhaps you could clarify exactly what this aims to achieve for all users of I2C.

jimmo, 2017-12-10:
It's possible my specific issue can be addressed with just a code change in drivers/display/ssd1306.py and hopefully I'm missing an obvious way to do this using the existing API. Either way, the issue I'm raising is exactly that: the supplied ssd1306 driver's `SSD1306_I2C` constructor fails in a confusing way if the user happened to create a hardware I2C instance.

peterhinch, 2017-12-11:
OK, I follow and can replicate this. As you say, when using hard I2C, the official driver calls unsupported I2C methods, namely `start()`, `stop()` and `write()`. These are documented as being available on ESP8266 only, but evidently work on the Pyboard provided that the I2C object is instantiated as soft I2C.

I agree that the solution is to extend the availability of these methods to Pyboard hardware I2C objects by modifying `i2c.c`. The way ahead is for you to submit a pull request once you're happy with your patch.

There won't be an issue with dependency on firmware versions so long as your patch doesn't affect existing code. Extending the availability of these methods to hard I2C instances shouldn't do this if it is done correctly. Clearly you'll need to test this.

jimmo, 2017-12-11:
It'd be really nice to not have to use the start/stop/read/write API... It puts a lot more burden on the user (e.g. getting the I2C addressing right -- look at ssd1306.py and see how `write_cmd` uses `self.addr` whereas `write_data` uses `self.addr << 1`).

```
def write_data(self, buf):
    self.i2c.writeto(self.addr, (bytes((0x40,)), buf,))
```

Or another suggestion -- writeto allows you to optionally specify stop=False. If it also allowed you to pass None (or zero) as the addr, then it could resume an existing write. That's a very simple and non-breaking change.

```
def write_data(self, buf):
    self.i2c.writeto(self.addr, bytes((0x40,)), stop=False)
    self.i2c.writeto(None, buf)
```

Alternatively, as suggested, a small improvement to `start()` (optionally taking the address and read/write flag) would greatly improve the usability of the start/stop/read/write API.

My question about firmware dependency from drivers is assuming new feature. Is there a particular version of uPy firmware that all code (or any specific file) in micropython/drivers is expected to work on?  (i.e. depending on either of the above changes to `writeto()` or the new version of `start()`)

dpgeorge, 2017-12-11:
I agree that there is a problem here, which really stems from the fact that not all hardware can support the I2C start/stop/read/write API as it's currently defined.  For example, nRF51 has pretty strict requirements for how the hardware TWI block is used: START must be directly followed by an address, and a repeated START only works if you inform the hardware when the byte before the repeated START is sent out.  So for the nRF51 it's only possible to implement raw start/stop/read/write commands using software/bit-banging.

To solve the issue here, it would be good to solve it in general rather than just stm32.  From an API point of view the choices seem to be:
- declare start/stop/read/write as methods that all I2C implementations should have and use bit-banging for archs that can't do it in hardware
- declare start/stop/read/write as *optional* methods and don't use them in portable drivers

For the second option the question is then how to deal with I2C writes (or reads) that are made from a combination of buffers, like in the SSD1306 driver.  Creating a temporary buffer to combine the smaller buffers is not really an option (allocates lots of RAM on the heap which is to be avoided).  So the readfrom/writeto API would need to be extended to allow such a segmented set of calls.

@jimmo you have some good suggestions how to implement the last point here (scatter/gather version of writeto, or allow to pass None as the address) but the things to consider are 1) is it too much of a burden to impose such an API on all ports; 2) can all ports even implement such an API?

> What is the policy on making code in micropython/drivers depend on newer firmware versions?

In general the changes to firmware try to be as backwards compatible as possible (so existing drivers, including 3rd party ones, still work).  On the other hand, a .py driver in this repo can change and users are expected to use the latest firmware for it to work.  So the latest drivers should work with the latest firmware (and usually old drivers will also work with the latest firmware).

jimmo, 2017-12-11:
Thanks @dpgeorge 

Can you think of any reason why a port wouldn't be able to implement either of these options? I can imagine the second option (addr=None) might be challenging if you had some sort of DMA-only I2C peripheral that didn't let you control the start and stop signal. (Does such a thing exist?)

I'm happy to make a PR that does any any of the discussed options (or combination) -- what would you prefer?

Thanks for the clarification about backwards compatibility. Sounds like we'd be fine in this case then for any of the options suggested.

dpgeorge, 2017-12-13:
> Can you think of any reason why a port wouldn't be able to implement either of these options?

Some ports (like Zephyr RTOS) don't have direct control over the hardware but rather go via a pre-defined API.  For Zephyr, the I2C driver doesn't allow individual STOP/START conditions, but it does provide (similar to what you suggested) a way to specify multiple chunks of the transaction in one call, via `i2c_transfer()`.  See eg https://github.com/zephyrproject-rtos/zephyr/blob/master/include/i2c.h#L195-L205 for how that function is used to implement a simple write.  That seems pretty powerful/general, but I still don't think it allows to support the proposed `addr=None` option (i2c_transfer looks like it always starts with the address).

So probably allowing an iterable of bytes objects would be the way that is the most compatible with ports.  (But note that it's not sensible to try and support every conceivable I2C hardware/API/interface so there needs to be a line drawn at some point.)

> I'm happy to make a PR that does any any of the discussed options (or combination) -- what would you prefer?

I guess you need the fix ASAP... so the path of least resistance (in terms of defining a new API and then implementing it across all ports) would be to make stm32 support start/stop/read/write for hardware I2C instances, maybe just by switching to a software implementation (just change the GPIO, use existing SW I2C code, then change the GPIO back to alt func)?

pfalcon, 2017-12-13:
> declare start/stop/read/write as optional methods and don't use them in portable drivers

They were never conceived as anything else than that. They are there to provide a proper layering implementation, like, *any* more higher-level I2C API can be implemented in terms of them, and they in turn can be implemented in terms of GPIO API. Thus, any port which provides GPIO, would automatically get I2C support, any port which provides start/stop/read/write (perhaps optimized beyond just GPIO bitbanging) can still implement any higher-level I2C API.

But a port not providing those, would be caught in a straitjacket of a particular higher-level I2C API, which might not work for a particular port (or require extensive workarounds to implement). And selecting a higher-level API which would suit any hardware was an open question (though the idea to follow Linux API model was proposed more than once - they've got to have done their homework to support as diverse HW as possible).

dpgeorge, 2019-05-20:
Fixed in 02afc0d241dbd0efb985ba5a7ded9cea616ffcd7

closed 2019-05-20 by dpgeorge