micropython/micropython #3484: naked attribute on nlr_push - Only basic asm statements can safely be included in naked functions

adritium, 2017-12-11:
One example:

https://github.com/micropython/micropython/blob/d3f82bc42576ccd80206a17a4941fd5c28c56530/py/nlrthumb.c#L70-L81

naked
[This attribute allows the compiler to construct the requisite function declaration, while allowing the body of the function to be assembly code. The specified function will not have prologue/epilogue sequences generated by the compiler. Only basic asm statements can safely be included in naked functions (see Basic Asm). While using extended asm or a mixture of basic asm and C code may appear to work, they cannot be depended upon to work reliably and are not supported.](https://gcc.gnu.org/onlinedocs/gcc/ARM-Function-Attributes.html#ARM-Function-Attributes)

`return 0` must be removed from all `naked nlr_push` functions.

dpgeorge, 2017-12-13:
> return 0 must be removed from all naked nlr_push functions.

As mentioned in the code, it's only there to silence compiler warnings.

adritium, 2017-12-14:
@dpgeorge thing is, it's causing the compilation to FAIL on clang 

https://forum.micropython.org/viewtopic.php?f=3&t=4017&p=23360&hilit=clang#p23360

SpotlightKid, 2017-12-14:
@adritium: Thanks for opening this issue. I meant to do this myself, but other things took precedence.

As things stand now, micropython and clang do not go together.

dpgeorge, 2017-12-14:
> it's causing the compilation to FAIL on clang

Using clang 5.0.0 on an x86-64 arch I don't see any issues compiling py/nlrx64.c, but I guess it could be a problem with other clang versions on other archs.

> If negating a gcc compiler warning causes a clang compiler failure, I don't think the trade-off is correct,

It's not a trade off, both things should be properly fixed.  Is there a suggestion how to fix it?

SpotlightKid, 2017-12-14:
>  but I guess it could be a problem with other clang versions on other archs.

Yes, I encountered this on armv7 (Samsung Galaxy S5 min and Raspberry Pi 3). Arm64 worked fine.

adritium, 2017-12-15:
@dpgeorge what I don't understand is why there would be a compiler warning in the first place if the `gcc` docs say **don't do this**?

What is the warning?

Maybe there's a bug in the `gcc` warning code?

dpgeorge, 2017-12-18:
> What is the warning?
> 
> Maybe there's a bug in the gcc warning code?

I agree it's inconsistent for gcc to say only use basic asm in naked functions, and then give a warning about not returning a value.  Revisiting the code now, removing the return statement from the naked nlr_push() function doesn't induce a warning.  So either a newer version of gcc fixed the issue, or the return was added incorrectly (maybe before the function became naked).

Anyway, I've attempted to fix it all properly in #3492, and in doing so take the chance to factor all the common NLR code into one place to clean up the individual machine-specific implementations.  Please take a look at that PR to see if it solves the issue.

dpgeorge, 2017-12-20:
Fixed by 6a3a742a6c9caaa2be0fd0aac7a5df4ac816081c

closed 2017-12-20 by dpgeorge