micropython/micropython #3486: WEBRPL

DanielIbaseta, 2017-12-12:


dpgeorge, 2017-12-13:
Nothing has changed on the http://micropython.org/webrepl/ page so it should be working as before.  Maybe it's the browser you are using?

dpgeorge, 2018-03-07:
I assume this was resolved.  If not feel free to reopen the issue with some further details about the problem.

closed 2018-03-07 by dpgeorge

jeremyherbert, 2018-06-02:
I think I know what this issue was. Latest chrome forbids a non-secure websocket from being opened from a secure page. The micropython website hosts webrepl via a https, so this is blocked. Perhaps a message could be added to the webrepl page that you need to save it to disk and run it from there?

```
VM979:164 Mixed Content: The page at 'https://micropython.org/webrepl/?' was loaded over HTTPS, but attempted to connect to the insecure WebSocket endpoint 'ws://192.168.0.49:8266/'. This request has been blocked; this endpoint must be available over WSS.
WrappedWebSocket @ VM979:164
connect @ (index):137
button_click @ (index):123
onclick @ (index):50

VM979:164 Uncaught DOMException: Failed to construct 'WebSocket': An insecure WebSocket connection may not be initiated from a page loaded over HTTPS.
    at new WrappedWebSocket (<anonymous>:164:21)
    at connect (https://micropython.org/webrepl/?:137:10)
    at button_click (https://micropython.org/webrepl/?:123:9)
    at HTMLInputElement.onclick (https://micropython.org/webrepl/?:50:92)
```

dpgeorge, 2018-06-04:
> The micropython website hosts webrepl via https, so this is blocked.

The website hosts both http and https, so visiting it from http://micropython.org/webrepl should allow the unsecured websocket without warning.

jeremyherbert, 2018-06-04:
Ah. Searching for "micropython webrepl" on google always links you to the https (at least for me)

peterhinch, 2018-06-04:
Perhaps a Google policy to deprecate http. DuckDuckGo does likewise.