micropython/micropython #3488: Add #ifdef __cplusplus extern "C" to all .h to enable calling from C++

adritium, 2017-12-13:
If you want to call any `mp*` function from C++, you have to tell the C++ compiler to use C-naming convention (i.e. no name mangling).

This would require adding the following to all micropython header files

```
#ifdef __cplusplus
extern "C" {
#endif

// mp function declarations

#ifdef __cplusplus
}
#endif
```

Since calling from C++ is a common enough workflow, I recommend we implement and merge this into master branch.

dpgeorge, 2017-12-22:
As noted before, uPy is already being used successfully in multiple C++ projects.  The way to include the headers in C++ code is to simply put the `extern "C" {` wrapper code at the location in the C++ where it includes the uPy headers.

OTOH, having these wrappers in C headers is very common and does make C++ integration easier.  It requires updating not only py/*.h files but also lib/*.h and extmod/*.h.  We can do this if there's enough interest from the community.

adritium, 2017-12-22:
I didn't realize you could put the `extern "C"` outside of the `#include` statements; that's good enough for me.

Are there any special cases doing it this way?

dxxb, 2017-12-22:
> OTOH, having these wrappers in C headers is very common and does make C++ integration easier. It requires updating not only py/.h files but also lib/.h and extmod/*.h. We can do this if there's enough interest from the community.

I for one sincerely wish and hope this does not ever happen 😃 Burden associated with the usage of a different language (in this case C++) by the user of a code base should not be transferred on the upstream code base.

dxxb, 2017-12-22:
> Are there any special cases doing it this way?

Not that I know of, it always worked as expected when I used it.

stinos, 2017-12-23:
Indeed it's the standard way of using a C api.  And for convenience you cold do it only once in a common header see https://github.com/stinos/micropython-wrap/blob/master/detail/micropython.h for example

closed 2018-04-17 by adritium