micropython/micropython #3490: Allow tools/make-frozen.py to accept folder OR multiple .py files

adritium, 2017-12-14:


chrismas9, 2017-12-14:
I will get back to #3340 after #3370. #3340 is primarily a way to parse symlinks under Windows. A symlink looks like a text file with a path/filename in it. As an extension I aso process a text file with multiple path/filenames in it.

I was then going to propose splitting the module directory into two, one for the port and one for the board. For STM32 I would put onewire.py in stm32/modules but put lcd160cr.py in boards/pybvxx/modules since it's not relevant to most STM32 boards.

For a custom project you could copy the board directory to MY_PROJECT, put your scripts in MY_PROJECT/modules and just "make BOARD=MY_PROJECT"  to get your frozen build.

Would this do what you want?

adritium, 2017-12-14:
I updated it with more detail
> `tools\make-frozen.py` requires a folder which it then recursively scans for all `.py` to freeze.

I'd like the option to give it one (or more) specific `.py` files and not need to place those `.py`s in a folder.

This is just a cosmetic request.

robert-hh, 2017-12-17:
If you like that behavior, why don't you implement it and make a PR? It is just a simple Python script. Please take care to keep it backward compatible.

dpgeorge, 2020-06-09:
Frozen code is now managed by a manifest file (see eg e81f538e2509938b25eb48ed2670142eed1f2573).  Feel free to open a PR adding support to manifest files for the feature mentioned here.

closed 2020-06-09 by dpgeorge