micropython/micropython #3491: Remove unsupported usage of __attribute__((naked))

saleemrashid, 2017-12-15:
At https://github.com/micropython/micropython/blob/63644016669c173190c71d39fb33ad9502fe2b0f/py/nlrthumb.c#L95 there is a function marked `__attribute__((naked))` containing a mixture of ASM and C code.

To quote https://gcc.gnu.org/onlinedocs/gcc/ARM-Function-Attributes.html#ARM-Function-Attributes

> While using extended asm or a mixture of basic asm and C code may appear to work, they cannot be depended upon to work reliably and are not supported.

adritium, 2017-12-15:
This is already addressed in #3484 
The function needs to be naked. What it also needs is for C-statements to be removed.
THAT would satisfy the `gcc` docs.

saleemrashid, 2017-12-15:
@adritium I did not mean to suggest removing `__attribute__((naked))`, I meant to remove the unsupported usage. Sorry if that was not clear :blush:. Also, I did not see your issue (I came here from trezor/trezor-core#89) so I will close my duplicate issue.

closed 2017-12-15 by saleemrashid