micropython/micropython #3498: Error in nlrthumb.c: 'noreturn' function does return when cross-compiling

TalSkverer, 2017-12-20:
Hi,
I'm cross compiling micropython on my Ubuntu (12.04) for an embedded system running Linux (armhf) and I ran into an error:

Looks like my compiler is upset that the `nlr_jump_tail` function is declared using the NORETURN attribute but still returns an error, while `nlr_push` is supposed to return an `unsigned int` but doesn't (even though eventually in the control flow there is an integer in r0. Weird.

My `arm-linux-gnueabihf-gcc` version is 4.6.3 (Old, I know, but it is a requirement for the target system)

I've searched through the recent issues for a solution and seen that this change to `nlrthumb.c ` has indeed been done [recently](https://github.com/micropython/micropython/issues/3484).

Here's the full error:
```
tal@tal:~/micropython/ports/unix$ PKG_CONFIG_PATH=~/micropython/lib/libffi/build_dir/ make CROSS_COMPILE=arm-linux-gnueabihf- V=1
python ../../py/makeversionhdr.py build/genhdr/mpversion.h
CC ../../py/nlrthumb.c
arm-linux-gnueabihf-gcc -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild -I../../lib/mp-readline -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_USSL=1 -DMICROPY_SSL_AXTLS=1 -I../../lib/axtls/ssl -I../../lib/axtls/crypto -I../../lib/axtls/config -DMICROPY_PY_BTREE=1 -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 -I../../lib/libffi/build_dir/out/lib/libffi-3.99999/include -DMICROPY_PY_FFI=1 -Os -fdata-sections -ffunction-sections -DNDEBUG  -U _FORTIFY_SOURCE -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMPZ_DIG_SIZE=16  -Os -c -MD -o build/py/nlrthumb.o ../../py/nlrthumb.c
../../py/nlrthumb.c: In function ‘nlr_jump_tail’:
../../py/nlrthumb.c:110:1: error: ‘noreturn’ function does return [-Werror]
../../py/nlrthumb.c: In function ‘nlr_push’:
../../py/nlrthumb.c:77:1: error: control reaches end of non-void function [-Werror=return-type]
cc1: all warnings being treated as errors
make: *** [build/py/nlrthumb.o] Error 1
```

Thanks in advance.

dpgeorge, 2017-12-22:
As you found, there was a very recent patch which re-introduced this issue.  It looks like old gcc versions have a bug where they incorrectly issue these warnings about return values (see eg https://patchwork.ozlabs.org/patch/173103/).

There are a few ways to fix this.  But the cleanest is probably to add `__builtin_unreachable();` at the very end of these 2 functions (nlr_jump_tail and nlr_push).

@TalSk can you confirm (or not) that this fixes your case?

TalSk, 2017-12-22:
@dpgeorge Indeed it has! (Even though now mpy_cross linking fails, but I'll look into it when I have some spare time). Thanks a bunch for your aid :)

closed 2017-12-22 by TalSk

dpgeorge, 2017-12-28:
This should be fixed by 97cc48553828ed091325d0d3922eb4cd6c377221