micropython/micropython #3506: Replacing SystemClock_Config() in ports/stm32/system_stm32.c

peter-conalgo, 2017-12-20:
To support my specific board (which is a product, not an eval board), I need to replace the clock setup code in`ports/stm32/system_stm32.c`, function `SystemClock_Config()`.

My changes are specific to my board, and cannot be expressed with the existing configuration variables for clock setup. It doesn't seem useful to change the port itself to support my board directly. Instead, I'd like to completely replace the `SystemClock_Config()` function with one suited to my board. I'm sure other boards would want to do that at times.

For now, I've made the linkage on this function weak (MP_WEAK) and overridden it elsewhere. However, this still requires the unused code to be compiled (so config values that are incorrect must be specified). 

How should I accommodate boards that need this sort of thing? I will publish the PR to achieve it.

dpgeorge, 2017-12-22:
The stm32 start-up code really needs a bit of cleaning up, to make it more efficient and less reliant on the ST HAL.  It also needs a few more start-up hooks for boards to initialise things at various points in the start up (MICROPY_BOARD_EARLY_INIT() already exists but it needs others at other locations).

Until then, please just continue to make the function weak and override it.

dpgeorge, 2021-02-02:
There are a lot of MICROPY_BOARD_xxx hooks that have been added recently, see 4c3976bbcaf58266fdcaab264fee5b7a94a682e5

So it might make sense now to add a similar option like `MICROPY_BOARD_SYSTEM_CLOCK_INIT` that by default calls `SystemClock_Config()` but can be overridden.

dpgeorge, 2022-02-03:
It was made a weak symbol in ae5f647a2d294cf940367918fbb250f1a9c0c8c0.

In the future configurability of this function can be improved if needed.

closed 2022-02-03 by dpgeorge