micropython/micropython #3507: mp_deinit() is commented out

adritium, 2017-12-20:
https://github.com/micropython/micropython/blob/a1d85d6199c03871ccf16700743160fad64d057b/py/runtime.c#L130-L138

Did something break that required this to be commented out?

stinos, 2017-12-20:
This happened in commit 1a1d11fa32ba043d22995d28cbc039cfa5f3cc46 (found via git blame/log) but without explicit comment as to why it happened - I'm assuming it has to do with reloading of modules. Not sure though. @pfalcon ?

dpgeorge, 2017-12-22:
> Did something break that required this to be commented out?

I don't think so.  I think the reason nothing is in mp_deinit() is because it's expected that you'll call mp_init() to re-init the system (unless the program exits completely), in which case there's no point in doing anything in mp_deinit().

adritium, 2017-12-22:
So should `mp_deinit()` be removed? Or at least comment should be added to explain why it's (almost) empty?

dpgeorge, 2021-04-30:
The commented out code has been removed in 6b7c8d3e7283a6c94687ba7dcb8beb94c2dc9eef

closed 2021-04-30 by dpgeorge