micropython/micropython #3508: Support for Dual Out flash mode (Support for ESP8285)

maribu, 2017-12-20:
Hi,

the ESP8285 is basically and ESP8266 with 1MiB of flash integrated. Thus, the MicroPython ESP8266 port should support it out of the box. Sadly, the integrated flash can only be used in Dual Out mode according to its [data sheet](http://espressif.com/sites/default/files/documentation/0a-esp8285_datasheet_en.pdf). This behavior is also observed in the [Tasmota firmware](https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips).

The good thing about DOUT flash mode is, that it seems to be supported by every ESP82* device and flash combination, according to the [Tasmota Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips). Sadly, it seems to be also the [slowest](https://www.esp32.com/viewtopic.php?t=1250).

So it would be nice to be able to configure the build to use a specific flash mode. The `FLASH_MODE` variable in the Makefile is only used in the `deploy` target so far. (By the way, recent versions of the esptool don't provide the `--flash_mode=` parameter any more.) Maybe this variable could be used to control which flash mode micropython is using?

Kind regards,
Marian

dpgeorge, 2017-12-22:
> Thus, the MicroPython ESP8266 port should support it out of the box. Sadly, the integrated flash can only be used in Dual Out mode

Yes it should work out of the box.  Did you try it?  MicroPython doesn't need to know the flash mode to execute because it calls the underlying system functions to access the flash.  The only time you need to specify that it's DIO is when flashing.

maribu, 2017-12-22:
Thanks for the fast reply. Yes, in fact I was unware my hardware was using an ESP8285 instead of a ESP8266 at first. Neither MicroPython nor NodeMCU is working, so I started to dig deeper and figured out recently sold products of the hardware in question are using the ESP8285 instead.

Btw, the more recent versions of esptool don't allow to manually specify the flash mode anymore. Obviously, they either detect the correct flash mode automatically or use DOUT as common dominator. (The esptool works fine and is able to upload and verify firmwares.)

The hardware in question is an itead sonoff basic. Using the [espurna](https://bitbucket.org/xoseperez/espurna) firmware works fine, so I rule out a hardware issue.

Kind regards,
Marian

peterhinch, 2017-12-22:
@maribu For general comments on the Sonoff you might be interested in [this forum thread](https://forum.micropython.org/viewtopic.php?f=16&t=3777&hilit=sonoff).
tl;dr I spent a considerable amount of time trying to get reliable operation from these and failed. Code which worked for days (until interrupted) on the reference hardware was unreliable on the Sonoffs.

maribu, 2017-12-22:
@peterhinch: Thanks for the input. Obviously you had an older revision using the ESP 8266 instead the ESP8285, so booting worked for you. If I had to guess what the problem was causing the unreliability, I would put my bet on a faulty device. The community behind the ESPurna are using these devices quite successful, so it shouldn't be a common problem (or MicroPython related, but your sonoff used a of the shelf ESP8266, which seems to work fine with MicroPython). Maybe you still have warrenty on your device and a backup of the original firmware to restore?

peterhinch, 2017-12-22:
@maribu This is best continued in the forum thread.

In essence I have four of these devices none of which are reliable running MicroPython networking code, although they will run code which doesn't use the WiFi. The same (WiFi) code running on the reference hardware under identical test conditions is reliable.

I'm hanging onto the devices in the hope that someone else manages to resolve the issue because they are a useful hardware package. In my view there are reasons for suspecting a hardware design issue, but equally good reasons for rejecting that explanation. As you say there is a community of users. Further I studied the schematic and tested the hardware; I couldn't find any issues. However I have no means of testing for effects resulting from the radiated RF.

It's possible the ESP8285 version is OK. I thought it worth issuing a note of caution, but do report your findings in the forum.

maribu, 2017-12-23:
Ok, my bad. It works out of the box when supplying the `--flash_mode=dout` parameter.

The esptool stopped listing all available options when calling `esptool.py --help`. Running `esptool.py <CMD> --help` lists the available options for the given command. 

So, the third byte of the firmware (counting from address 0x00) configures the [flash mode](http://blog.zazolabs.com/esp8266-using-different-flash-sizes-fota-download-tool/). It was - at least to me - very counter intuitive that `esptool.py` does patch the uploaded image on the fly when supplying the `--flash_mode=` option. The reason for the ESPurna firmware to work out of the box without the `--flash_mode=` parameter is, that the firmware was created to use `DOUT`.

May I suggest to change the Makefile as follows:

``` patch
diff --git a/ports/esp8266/Makefile b/ports/esp8266/Makefile
index 9d6e502c..efc2c023 100644
--- a/ports/esp8266/Makefile
+++ b/ports/esp8266/Makefile
@@ -179,7 +179,7 @@ reset:
 
 $(FWBIN): $(BUILD)/firmware.elf
        $(ECHO) "Create $@"
-       $(Q)esptool.py elf2image $^
+       $(Q)esptool.py elf2image --flash_mode=$(FLASH_MODE) $^
        $(Q)$(PYTHON) makeimg.py $(BUILD)/firmware.elf-0x00000.bin $(BUILD)/firmware.elf-0x[0-5][1-f]000.bin $@
 
 $(BUILD)/firmware.elf: $(OBJ)
```

This would create the image using the flash mode configured in the `Makefile`. Thus, deploying the image without the `--flash_mode` parameter will work out of the box.

Kind regards,
Marian

maribu, 2017-12-23:
In fact the [wiki](http://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/intro.html#deploying-the-firmware) states this. But it would be nice to add a sentence to say that the ESP8285 requireds `--flash_mode=dout` to operate.

closed 2017-12-23 by maribu

mindforger, 2018-05-08:
okay, sorry for disturbing this aged issue but it fit best, but i don't have a forum account.

can i use the wiki with said parameter to flash a Wemos D1 with a default ESP8266 binary?

dpgeorge, 2018-05-08:
> can i use the wiki with said parameter to flash a Wemos D1 with a default ESP8266 binary?

Yes you should be able to use dio mode.