micropython/micropython #3510: uselect missing from ESP32 port

stroobandt, 2017-12-21:
Could it be that `uselect` is missing from the ESP32 port, in analogy with https://github.com/micropython/micropython/issues/2380 ?

That would be a real pity, because `uasyncio` is depending on it.
```
Traceback (most recent call last):
  File "main.py", line 6, in <module>
  File "/flash/lib/uasyncio/__init__.py", line 2, in <module>
ImportError: no module named 'uselect'
```

robert-hh, 2017-12-21:
try "import select"

dpgeorge, 2017-12-22:
It should be there.  You can do `help('modules')` to get a list of built-in modules.

stroobandt, 2017-12-22:
Sorry about that. As a newbee, I am still confused about when to use the `u` prefix in module names.
I got the following error when trying out the `uasyncio` module:
```
File "/flash/lib/uasyncio/__init__.py", line 2, in <module>
ImportError: no module named 'uselect'
```
Neither was `upip` of any help:
```
$ ./micropython -m upip install uselect
Installing to: /home/serge/.micropython/lib/
Warning: pypi.python.org SSL certificate is not validated
Error installing 'uselect': Package not found, packages may be partially installed
```

stroobandt, 2017-12-22:
I opened a `uasyncio` bug here: https://github.com/micropython/micropython-lib/issues/241

closed 2018-03-07 by dpgeorge