micropython/micropython #3512: uctypes pointer on 64bit platforms not 64bit

ryannathans, 2017-12-22:
```
./micropython
MicroPython v1.9.3-7-g46e9ef6-dirty on 2017-12-21; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import uctypes
>>> definition = {'test': uctypes.PTR}
>>> buffer = bytearray(100)
>>> struct = uctypes.struct(uctypes.addressof(buffer), definition)
>>> struct.test = uctypes.addressof(buffer)
>>> struct.test
3253563904
>>> uctypes.addressof(buffer)
140410029437440
>>>
```

dpgeorge, 2017-12-22:
According to the docs this is not how PTR is intended to be used.  Instead you must specify the type of the pointed-to-memory, (eg pointer to UINT8 would be `{'test': (uctypes.PTR, uctypes.UINT8)}`).  Please see the docs: http://docs.micropython.org/en/latest/pyboard/library/uctypes.html?highlight=uctypes#defining-structure-layout

ryannathans, 2017-12-22:
```
>>> buffer = bytearray(100)
>>> new_buffer = bytearray(10)
>>> definition = {'test': (uctypes.PTR, uctypes.UINT8)}
>>> struct = uctypes.struct(uctypes.addressof(buffer), definition)
>>> struct.test = uctypes.addressof(new_buffer)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: syntax error in uctypes descriptor
>>> struct.test = 5
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: syntax error in uctypes descriptor
>>>
```

Following the documentation, this use of uctypes.PTR seems broken - unless I'm doing something wrong still? Looks like it doesn't understand the dictionary definition passed in.

dpgeorge, 2017-12-22:
You can't update the pointer (ie `struct.test`), only the values it points to (ie `struct.test[0]`).  To change the pointer you must change the underlying buffer containing the pointer value, in this case the first 8 bytes of `buffer`.

closed 2018-01-15 by ryannathans