micropython/micropython #3515: Any plans to support the new H7 series ?

iabdalkader, 2017-12-25:
Hi, is anyone currently working on the new H7 series ? I have some working code (on a custom board, still waiting for the dev boards to be released) and I figured I should ask first. Do you have plans to support it ? If not, do you mind if we send in patches to support it at some point in the near future ? Note the H7 is quite different than the F7 (there are so many changes to HAL) so it's going to need a lot of changes to drivers.

dpgeorge, 2017-12-28:
Yes I do plan to support the STM32H7xx.  Feel free to submit patches for it, although it would need to be done in a clean/minimal way.  To start with the CMSIS files should be added to the stm32lib submodule.

iabdalkader, 2018-01-03:
Thank you, I will start by sending HAL and CMSIS patches as soon as I get the dev boards.

iabdalkader, 2018-02-19:
@dpgeorge Should I add the board files next ?

rolandvs, 2018-02-19:
@iabdalkader Do you have it running already?

iabdalkader, 2018-02-19:
@rolandvs No not yet, I was just getting started. I added the HAL to https://github.com/micropython/stm32lib/commit/acaf64eb1e6084495cebb716eac425fdef155b61 and it's merged. Need any help ?

rolandvs, 2018-02-19:
It starts compiling :-), however working my way through the ```#error unsupported processor``` first and then I'm going through it again looking for all ```#ifdef``` stuff concerning the various families. For now the system_stm32.c is the most important to get it running, etc. See what happens. In coming days I update my fork with my efforts... So let's help each other :-)

iabdalkader, 2018-02-19:
@rolandvs Maybe you should send the board files first, and then assign issue or something.

rolandvs, 2018-02-19:
@iabdalkader Always get a bit nervous with git. For myself I would prefer not to bother the main repo with my ramblings until it is sort of running and worthy for a PR. So I created a NUCLEO-H743ZI branch at my fork and put the board stuff there. This is the easy part of the porting... :-) So let me know how to proceed otherwise...

iabdalkader, 2018-02-19:
@rolandvs The HAL has changed a lot I don't think you can add support for H7 in a single PR, but it's not up to me. @dpgeorge any thoughts ?

dpgeorge, 2018-02-20:
> The HAL has changed a lot I don't think you can add support for H7 in a single PR, but it's not up to me. @dpgeorge any thoughts ?

Let's try and get something minimal compiling and working first, even if it just has a UART REPL with no support for peripherals.  And then we can merge the required changes.

Recently I've been cleaning up the stm32 board config so it's more flexible, and adding the ability to disable more stuff.  We can keep going in this direction to get the H7 working, and then gradually add in peripheral support.

To help make progress here, in #3624 I've made a PR to add the af.csv file for the H7, with the list of alt funcs on the GPIO.

@iabdalkader perhaps you can make your own PR and push to it all the changes needed to get the H7 compiling.  Then we can discuss them better.

iabdalkader, 2018-02-20:
@dpgeorge I'm working on it. BTW the stm32lib submodule is not updated with the H7 HAL.

dpgeorge, 2018-02-20:
> BTW the stm32lib submodule is not updated with the H7 HAL.

Yes that's true.  The new H7 tag needs to be made into a new working branch with the existing custom mods applied, then it can be pulled into this repo.  But let's wait until the H7 is working before doing that.  At least with the H7 stuff in stm32lib it's possible to update the submodule yourself locally to test it out.

iabdalkader, 2018-02-21:
@dpgeorge I updated the submodule. Can we add a feature to disable the ADC ? I don't want to have to change a driver unless it's absolutely necessary. So far I'm getting error in dma.c, flashbdev and adc.c

dpgeorge, 2018-02-22:
> Can we add a feature to disable the ADC ?

Ok, added in 60b0982bb2d82b1c7b026a6f9e227e6dc837b005

> The H7 HAL_FLASH_Program() writes 256-bits flash words. Should I change the HAL to write 32 bits or pad the data to 256 bits ?

There's likely a reason it must program 256 bits at a time, so we can't change the HAL.

Our flash_write() function is always called to program an entire page so you can assume the length is aligned to 32 bytes (256 bits) and then there's no need to pad.

iabdalkader, 2018-02-23:
> There's likely a reason it must program 256 bits at a time, so we can't change the HAL.

I don't think so it just writes two double words in a loop. Double-word, word, half-word and byte write operations are supported (according to the datasheet). They probably just didn't want to implement all the programming functions.

Also this line in CMSIS stm32h743xx.h:2355 has to be commented.

`#define SDMMC               ((SDMMC_TypeDef *) SDMMC_BASE)`

```
In file included from ../../lib/stm32lib/CMSIS/STM32H7xx/Include/stm32h7xx.h:117:0,
                 from ../../lib/stm32lib/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_def.h:48,
                 from ../../lib/stm32lib/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_rcc.h:47,
                 from boards/NUCLEO_H743ZI/stm32h7xx_hal_conf.h:207,
                 from ../../lib/stm32lib/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal.h:48,
                 from ./mpconfigport.h:292,
                 from ../../py/mpconfig.h:45,
                 from ../../py/mphal.h:29,
                 from ../../py/mpprint.c:33:
../../lib/stm32lib/CMSIS/STM32H7xx/Include/stm32h743xx.h:2355:45: error: expected ')' before '*' token
 #define SDMMC               ((SDMMC_TypeDef *) SDMMC_BASE)
                                             ^
./pin_defs_stm32.h:128:20: note: in expansion of macro 'SDMMC'
     void          *SDMMC; \
                    ^~~~~
./pin.h:46:5: note: in expansion of macro 'PIN_DEFS_PORT_AF_UNION'
     PIN_DEFS_PORT_AF_UNION
     ^~~~~~~~~~~~~~~~~~~~~~
../../lib/stm32lib/CMSIS/STM32H7xx/Include/stm32h743xx.h:2355:48: error: expected ')' before 'SDMMC_BASE'
 #define SDMMC               ((SDMMC_TypeDef *) SDMMC_BASE)
                                                ^
./pin_defs_stm32.h:128:20: note: in expansion of macro 'SDMMC'
     void          *SDMMC; \
                    ^~~~~
./pin.h:46:5: note: in expansion of macro 'PIN_DEFS_PORT_AF_UNION'
     PIN_DEFS_PORT_AF_UNION
     ^~~~~~~~~~~~~~~~~~~~~~
In file included from ./mphalport.h:3:0,
                 from ../../py/mphal.h:34,
                 from ../../py/mpprint.c:33:
./pin.h:47:3: error: no semicolon at end of struct or union [-Werror]
   };
```

dpgeorge, 2018-02-23:
> I don't think so it just writes two double words in a loop. Double-word, word, half-word and byte write operations are supported (according to the datasheet). They probably just didn't want to implement all the programming functions.

Ok, I see.  But right now I think the simplest thing to do is just make flash.c write 256 bits at a time.

> Also this line in CMSIS stm32h743xx.h:2355 has to be commented.

Yes that does pose a problem.  Maybe it's better to make all those union fields lower case in `#define PIN_DEFS_PORT_AF_UNION` to prevent future issues like this.

iabdalkader, 2018-02-23:
> Yes that does pose a problem. Maybe it's better to make all those union fields lower case in #define PIN_DEFS_PORT_AF_UNION to prevent future issues like this.

That should fix it without changing HAL/CMSIS, I just commented the line for now. Note `SDMMC_BASE` is not actually defined or used anywhere.

dpgeorge, 2018-02-23:
Let's try to collect all the patches required to get it compiling and running a REPL (maybe no filesystem if that's too difficult) and then see what the best way is to merge the patches.

dpgeorge, 2018-02-23:
Regarding DMA, it may be enough to just disable dma.c like this:
```diff
--- a/ports/stm32/dma.c
+++ b/ports/stm32/dma.c
@@ -32,6 +32,16 @@
 #include "dma.h"
 #include "irq.h"
 
+#if defined(STM32H7)
+volatile dma_idle_count_t dma_idle;
+void dma_idle_handler(int tick) {
+}
+void dma_init(DMA_HandleTypeDef *dma, const dma_descr_t *dma_descr, void *data) {
+}
+void dma_deinit(const dma_descr_t *dma_descr) {
+}
+#else
+
 typedef enum {
     dma_id_not_defined=-1,
     dma_id_0,
@@ -506,3 +516,5 @@ void dma_idle_handler(int tick) {
         }
     }
 }
+
+#endif
```

iabdalkader, 2018-02-23:
I will collect all the patches (if any) and report them after I send the PR. Note I just finished dma.c but I couldn't find the DMA1/2 request mapping in the datasheet (but they seem to be the same as F7) and I can't find any requests for the SDMMC (but the SDCARD is disable anyway). Also can't figure out the purpose of the DMAMUX1/2 and if they're really needed for the basic DMA (the examples for I2C and SPI don't use it). DMA should work, we'll see about that, and the board files, system_stm32 and filesystem are all done.

iabdalkader, 2018-02-24:
@dpgeorge Finally got an image and it seems to be working (tried calling flash_error in main) and there's some USB activity, but USB doesn't fully enumerate. I get multiple `device not accepting address 115, error -71` errors. Any ideas ?

dpgeorge, 2018-02-24:
> Finally got an image and it seems to be working

Great!  Did you manage to get a UART REPL working through the STLINK USB-serial adaptor?

> but USB doesn't fully enumerate. I get multiple device not accepting address 115, error -71 errors. Any ideas ?

USB is hard.  It could be that USB IRQs are not working.

iabdalkader, 2018-02-24:
> Great! Did you manage to get a UART REPL working through the STLINK USB-serial adaptor?

Yes, but I wanted to get USB working. Should I send the PR now ?

> USB is hard. It could be that USB IRQs are not working.

It's just weird because I got our USB/CDC bootloader working on OpenMV-H7.

dpgeorge, 2018-02-25:
> Should I send the PR now ?

If you send a PR then I can try to help to debug the USB issue.

> maybe a clock issue.

Make sure the HSE value is correct and the PLLs are set correctly.  You could try toggling a GPIO in a loop and measuring its frequency on a scope to understand the actual CPU frequency.

iabdalkader, 2018-02-25:
> Make sure the HSE value is correct and the PLLs are set correctly.

I used the values generated with STM32Cube.

iabdalkader, 2018-02-27:
Is anyone else having issues debugging NUCLEO_H743ZI on Arch ?

dpgeorge, 2018-03-09:
The STM32H7 is now supported.

closed 2018-03-09 by dpgeorge