micropython/micropython #3516: Round to the nearest float with the correct least significant decimal.

stroobandt, 2017-12-25:
You have all been very naughty!
Look at the following Christmas bug Santa has brought…

The problem concerns the [builtin function `round()`](http://docs.micropython.org/en/latest/pyboard/library/builtins.html?highlight=round#round).
In its current implementation, rounding is done to the nearest float that can be represented by the machine.
This is wrong, rounding should be done to the nearest float **with the correct least significant decimal** that can be represented by the machine.

Consider the following little `main.py` script:

    f = 1011.640071
    print('\n%f = float' % f)
    print('%f = float rounded to one decimal' % round(f, 1))
    print('%6.1f = formatted rounded float\n' % round(f, 1))

    delta = 0.000040
    print('%f = float correctly rounded to least significant first decimal' % (round(f, 1) + delta))
    print('%6.1f = formatted correctly rounded float\n' % (round(f, 1) + delta))

    epsilon = abs(7./3 - 4./3 - 1)
    print('%f = 1000 × machine epsilon' % (1000 * epsilon))
    print(epsilon)
    print()

Here is the output on a PyBoard:

    MicroPython v1.9.2-87-gda8c4c26 on 2017-09-13; PYBv1.1 with STM32F405RG

    1011.639952 = float
    1011.599898 = float rounded to one decimal
    1011.5 = formatted rounded float

    1011.600017 = float correctly rounded to least significant first decimal
    1011.6 = formatted correctly rounded float

    0.000119 = 1000 × machine epsilon
    1.192093e-07

Things are not any different on a Pycom LoPy ESP32:

    MicroPython 2ac6da2 on 2017-12-18; LoPy with ESP32

    1011.639952 = float
    1011.599898 = float rounded to one decimal
    1011.5 = formatted rounded float

    1011.600017 = float correctly rounded to least significant first decimal
    1011.6 = formatted correctly rounded float

    0.000119 = 1000 × machine epsilon
    1.192093e-07

Note that the difference between correct and erroneous rounding is about a 1000 times the machine epsilon, which makes sense.

For certain applications, such poor rounding can have very serious consequences down the line…

This finding may account for the multitude of other reports complaining about `round()` behaviour:
- https://forum.micropython.org/viewtopic.php?t=802
- https://github.com/bbcmicrobit/micropython/issues/367
- https://github.com/micropython/micropython/issues/2616
- https://support.microbit.org/support/solutions/articles/19000026098-strange-rounding-error

dpgeorge, 2017-12-28:
Thanks for the report.  Note that there are bugs in the float formatting routine (independent to the report here), and that can confuse the results when trying to print out rounded values.

What seems to be reported here is that `round(1011.640071, 1)` with 32-bit float is rounded "too low" and that there exists a closer machine value (representable in a 32-bit float) to the true 1011.6 value.

If you have ideas how to fix this then patches are welcome.

stroobandt, 2017-12-28:
Thank you for looking into this.
Float formatting by itself is not involved, but I would not be surprised that many of those bug reports stem from the problem reported here.

> What seems to be reported here is that round(1011.640071, 1) with 32-bit float is rounded "too low" and that there exists a closer machine value (representable in a 32-bit float) to the true 1011.6 value.

That is not entirely correct. `round(1011.640071, 1)` _is_ being rounded to the nearest representable 32-bit float, which happens to be `1011.599898`.

However, the second nearest representable 32-bit float `1011.600017` —which is actually still within the machine epsilon range— would be a better float representation simply because its least significant decimal `.6` is correct. When rounding, the least significant decimal of the float representation is what needs to be correct. This is the essential thing one is after when rounding. What follows this decimal does not matter.

I consider this to be a very serious bug, worthy of a new MicroPython release when resolved. In scientific experiments, this bug may cause formatted floats to suddenly jump from `1011.7` to `1011.5`, when the underlying float for the latter is actually closer to `1011.6`. This is how I noticed this. Errors of this magnitude are actually capable of sparking global warming climate discussions, to say the least.

I am willing to have a look at the C code, even though I am not experienced with C.
Could you provide a link to the actual routine?
I have been browsing this GitHub repository, but was unable to find it.

By the way, the MicroPython code for testing the `round()` builtin function would also better include less naive example values.

stroobandt, 2017-12-28:
I found [the C routine responsible for rounding floats](https://github.com/micropython/micropython/blob/58f00d7c0e116e17ab141ae18275edea14b88431/py/modbuiltins.c#L465).

dpgeorge, 2017-12-28:
> When rounding, the least significant decimal of the float representation is what needs to be correct

In that case, so I understand things better, does the follow example also illustrate the problem?  Note that this example is run on normal CPython (not MicroPython) using 64-bit floats:
```python
>>> '%.20f' % round(8000000000001.64, 1)
'8000000000001.59960937500000000000'
```
According to the issue here, the value should have been rounded to:
```python
>>> '%.20f' % (round(8000000000001.64, 1) + 5e-4)
'8000000000001.60058593750000000000'
```
Is that correct?

stroobandt, 2017-12-28:
Yes, Damien, that is exactly the issue.
You have beaten me by a couple of minutes!
I have also been running test with CPython on an amd64 platform and saw the same.
Apart from that, I have been reading up to the theory and standards.

I now have come to believe we are not dealing with a MicroPython bug, but rather with an issue in the [IEEE 754-2008 standard](https://en.wikipedia.org/wiki/IEEE_754) which happens to be [up for revision](https://en.wikipedia.org/wiki/IEEE_754_revision#correctly_rounded_base_conversion) and [open discussion](http://754r.ucbtest.org/) in 2018.

As said, the 2008 version of the [IEEE Standard for Floating-Point Arithmetic (IEEE 754)](https://en.wikipedia.org/wiki/IEEE_754) is what applies and both C and CPython are compliant with this. (I assume MicroPython by inference too.)

The problem is of course a classical binary to decimal conversion problem.
I found an excellent [theoretical description](http://www.validlab.com/goldberg/paper.ps) of this; see pages 236-237 (pp. 66-67 of the PostScript file).
There is also an excellent page discussing the [floating point arithmetic issues and limitations of CPython](https://docs.python.org/3/tutorial/floatingpoint.html)

> On most machines today, floats are approximated using a binary fraction with the numerator using the first 53 bits starting with the most significant bit and with the denominator as a power of two. In the case of 1/10, the binary fraction is `3602879701896397 / 2 ** 55` which is close to but not exactly equal to the true value of 1/10.

Larger platforms map Python floats to IEEE-754 “double precision”. 754 doubles contain 53 bits of precision. By consequence, most CPython floating point limitations go by unnoticed.

Due to hardware limitations, this is not the case with 32-bit MicroPython. The resulting limitations are depicted in the following graph.

[image]

stroobandt, 2017-12-29:
It seems that IEEE 754-2008 is [up for revision in 2018 about this very issue](https://en.wikipedia.org/wiki/IEEE_754_revision#correctly_rounded_base_conversion). Perhaps, we as the MicroPython community, should [chime in on this open revision process](http://754r.ucbtest.org/). The trade-off between a tiny loss in precision (still within one [ULP](https://en.wikipedia.org/wiki/Unit_in_the_last_place#Language_support)) versus gain in rounding convenience appears to be open for debate.

The following stopgap solution resolves the issue on all platforms, but it may look a bit counter-intuitive to the casual programmer:
```
>>> f = 1011.639952
>>> print('%f' % (round(f,1)))
1011.599898
>>> print('%s' % (repr(round(f,1))))
1011.6
```

The question is now: Who is at fault of this human-unfriendly behaviour; string formatting or ultimately IEEE 754-2008 through the `round()` function?
One thing is sure: A lot of people are falling into this rounding trap!

stroobandt, 2017-12-29:
Still, there seems to bing something wrong…
```
>>> f = 1011.60004
>>> print('%f' % f)
1011.600017
```
OK, but this should have the same internal representation:
```
>>> f = 1011.60003
>>> print('%f' % f)
1011.599898
```
However, it has an internal representation that is further off.

ddiminnie, 2019-04-12:
This is a fascinating conversation!  I thought I'd point out one thing, however:

Rounding a floating-point value to a specified number of digits past the decimal point is not specified in IEEE-754 (2008), nor is it specified in the C99 standard.  Those standards do specify how to convert values from one radix to another, as well as how to round a floating-point value to the nearest integer - neither of which gives a definitive way to handle the problem that python's 2-argument "round()" function addresses.

Because there isn't a specified method for performing rounding to a given number of decimal places, implementors desiring this functionality are left to devise their own methods.  One common method (which happens to be the method used by MicroPython) is as follows:
```
round(x, n) --> nearest_integer(x * 10**n)/10**n
```

For the cited example (where x == 1011.640071 and n == 1), this calculation proceeds as follows:

1. Convert the stream of decimal digits '1011.640071' to a binary32 value (in a perfect implementation): 
```0b1.11111001110100011110111 * 2**9 (== 1011.64007568359375)```

1. Multiply the binary32 equivalent of the input by ```10**1``` (exact multiplication rounded to 24 significant bits):
```0b1.00111100001000110011010 * 2**13 (== 10116.400390625)```

1. Compute the nearest integer to the product (exact):
```0b1.00111100001000000000000 * 2**13 (== 10116)```

1. Divide the integer result by ```10**1``` (exact division rounded to 24 significant bits):
```0b1.11111001110011001100110 * 2**9 (== 1011.5999755859375)```

1. Represent the answer in decimal using the recommended precision of 9 digits (of course, the external representation of the result doesn't really matter if the internal representation doesn't match the desired value...):
```1011.59998```

Naturally, the result above assumes that each step is performed perfectly (subject to rounding to the target precision).  As dpgeorge mentions, the decimal --> binary conversion in the first step isn't perfect in MicroPython.  Moreover, floating-point arithmetic may differ subtly from one platform to another, depending on the underlying implementation, making it harder to obtain consistent results across the various supported ports.

No standard exists for this type of operation because it is an ill-defined operation (since the internal representation of the input to the problem often is not exact, and since operations needed to carry out the rounding themselves may suffer from roundoff, and conversion of the result back to external representation is also often not exact).  I'm not sure there is a fix (in the sense of providing the behavior requested in the original description) that does not introduce incorrect results for other examples.

vinc17fr, 2019-04-12:
I don't think there is any bug, certainly not in the `round` function. But I don't know Python and MicroPython, so I'm not sure of the required behavior. Here's what happens. In `f = 1011.640071`, the value is converted to binary32 (single precision), which is about 1011.639952 according to the first output line. Then `round(f, 1)` rounds it to one decimal place, which is 1011.6, but for the second output line, this decimal value is converted back to binary32, which is about 1011.599898. Concerning the third output line, this binary32 value (which is about 1011.599898) is converted to one decimal place by the `%6.1f` format specifier. Here I'm not sure, but it seem that the rule is to truncate, hence the output `1011.5` (if the rule were to round to nearest, then I agree that there would be a bug, but the rounding rule would need to be confirmed).

mdickinson, 2019-04-12:
> But I don't know Python and MicroPython, so I'm not sure of the required behavior.

I can answer for Python (or at least, for CPython, versions >= 3.2). Officially, the language documentation refuses to make guarantees. But unofficially, on the vast majority of machines, CPython should be doing correct rounding (ties to even) for all binary <-> decimal conversions, including string formatting, parsing of numeric literals, etc. The one exception is the `str` or `repr` of a float value, which uses the now somewhat common "shortest string" representation. CPython uses a derivative of David Gay's `dtoa.c` for all these conversions (including the shortest string conversion), under the assumption that the machine is using IEEE 754 binary64 for its C `double`. (In the unlikely event that the floating-point format is _not_ IEEE 754 binary64, it uses system calls instead; these may or may not give correct rounding.)

If you witness CPython not doing correct rounding for such conversions, I'd consider that a bug that should be reported to bugs.python.org (and please add "mark.dickinson" to the nosy list if you make such a bug report).

From the CPython perspective, having "%6.1f" produce "1011.5" instead of "1011.6" from an input of `1011.599898` would be considered a bug. But this tracker is for MicroPython, not CPython, and the rules may be different.

mdickinson, 2019-04-12:
[This Stack Overflow answer](https://stackoverflow.com/a/22155830/270986) (disclaimer: I wrote it) may be helpful for explaining what `round` does on CPython.

vinc17fr, 2019-04-12:
The first part of your [StackOverflow answer](https://stackoverflow.com/questions/13479163/round-float-to-x-decimals/22155830#22155830) corresponds to the beginning of what I've explained about the example, and I agree with you concerning the rest of this answer. That said, the [MicroPython `round` implementation](https://github.com/micropython/micropython/blob/58f00d7c0e116e17ab141ae18275edea14b88431/py/modbuiltins.c#L460) is inaccurate, but this is not visible in this bug report (or not clearly visible, because I haven't checked closely).

For the `%6.1`, that's necessarily a different implementation, because `round` returns a binary32 value instead of a string. I assume that in this implementation, there's a truncate somewhere (that would explain the output, otherwise this would mean that the algorithm is very inaccurate, more than the one for `round`). Whether this is expected or not for MicroPython, I don't know.

vinc17fr, 2019-04-12:
Looking at the example more closely, that's even surprising that `1011.599898` is output. In C, using float (binary32), I get `1011.640076`. I suspect that various roundings are done toward zero.

vinc17fr, 2019-04-12:
A test can be done with [GNU MPFR](https://www.mpfr.org/):
```
#include <stdio.h>
#include <mpfr.h>

int main (void)
{
  mpfr_t d, f;
  char s[] = "1011.640071";

  mpfr_init2 (d, 30);
  mpfr_init2 (f, 24);
  mpfr_set_str (d, s, 10, MPFR_RNDN);
  mpfr_printf ("%Rb\n", d);
  mpfr_set_str (f, s, 10, MPFR_RNDN);
  mpfr_printf ("%.23Rb %Rf\n", f, f);
  mpfr_nextbelow (f);
  mpfr_printf ("%.23Rb %Rf\n", f, f);
  mpfr_nextbelow (f);
  mpfr_printf ("%.23Rb %Rf\n", f, f);
  mpfr_clears (d, f, (mpfr_ptr) 0);
  return 0;
}
```
One gets:
```
1.11111001110100011110110111011p+9
1.11111001110100011110111p+9 1011.640076
1.11111001110100011110110p+9 1011.640015
1.11111001110100011110101p+9 1011.639954
```
The first line shows the approximate binary value with more precision than binary32. The second line shows the value converted to the nearest binary32 value and a decimal approximation (corresponding to `%f`); when looking at the first line, one can see that the value has been rounded upward. The 3rd and 4th lines show the preceding values in binary32.

If MicroPython rounds 1011.640071 to nearest in binary32 (to about 1011.640076), then the `%f` conversion would lose around 2 binary ulps here. But if 1011.640071 is rounded toward 0 (to about 1011.640015), then the `%f` conversion would lose around 1 binary ulp.

yur-spiridonov, 2019-10-29:
Hello! I want to invite you to consider my algorithm for correctly rounding decimal numbers represented in binary code to a given number N significant decimal digits. Below is a test program for converting a positive X number from a floating point of any format to a floating-point number xr, whose decimal value is the number closest to the decimal number correctly rounded to N significant digits. Read more about this here.
At the entrance:
- number N <= 15 - the number of decimal digits to which rounding should be performed.
- x is any decimal positive number.
 The decimal number X is displayed on the console, the value of which is equal to double (x). The program rounds exactly the number X.

Returns:
- Xr - decimal number with integer mantissa correctly rounded to N significant digits.
- xr - decimal number in double format, closest to decimal number Xr.

#include <iostream>
#include <math.h>
#include <stdlib.h>
#include <iomanip>
using namespace std;

int main()
{
   double x,xr,X;
   unsigned long long int Xr;
   int N,E,e,k;

  cout <<"Input a decimal precision N=";
  cin>>N;
  cout <<endl<<"Input a number and press ENTER:"<<"\n"<<"x= ";
  cin>>x;

   cout<<"X= "<< setprecision(18)<<x << '\n';

    frexp (x, &e);
    E =static_cast <int> (e*0.301);
    k=N-E;
   if (E<0)       //for format xr=d0.d1...dN*10^E (d0≠0).
        k=k+1;
    X=x*pow(10,k);
       if (X > pow (10,N)){
            X=X/10;
            k=k-1;
      }

       Xr=static_cast <unsigned long long int> (X+0.5);
       xr=Xr*pow(10,-k);

    cout<<endl <<"Xr= "<<Xr<<"e"<<-k<<'\n';
    cout<<"xr="<<xr<<'\n';

      return 0;
}

yur-spiridonov, 2019-10-29:
Read more about this [here]([url](url)).

vinc17fr, 2019-10-29:
Your code does not take rounding errors into account, starting with `X=x*pow(10,k);`. Thus it won't work.

yur-spiridonov, 2019-10-29:
For some reason, when sending a comment, the multiplication sign disappears in X=x*pow(10,k); Try again.

yur-spiridonov, 2019-10-29:
Read [here](https://github.com/yur-spiridonov/Decimal-rounding-of-binary-numbers/commit/master)

yur-spiridonov, 2019-10-29:
Read more about this [here](https://github.com/yur-spiridonov/Decimal-rounding-of-binary-numbers/commit/master).

yur-spiridonov, 2019-10-29:
Could you give an example of such a hard-to-round case? Thank you.

vinc17fr, 2019-10-29:
I don't currently have hard-to-round cases, but one could write a program to find them. Here, this is a bit tricky due to the double conversion: as specified,
1. the float is first regarded as an exact decimal number,
2. then this decimal number is rounded to some position,
3. then this rounded value is converted to float (binary32).

The hardness to round must take into account both steps 2 and 3, which are both discontinuous functions. In general, when considering the Table maker's dilemma, one has a continuous function that is rounded (the rounding being a discontinuous function). But here, 2 rounding functions are involved, thus 2 discontinuous functions. This needs some thoughts to find/build hard-to-round cases...

yur-spiridonov, 2019-10-30:
Here x is an arbitrary non-representable decimal number. For example, x = 1.2345678901234e-32. X is the internal representation of the number x in double format. In our case, double(X) = **1.234567890**1233999e-032.
Information about the exact value of the number x is lost for us as a result of converting the number x to base 2. Therefore, we round double (X). If N = 10, then the exact value of the rounded number X is equal to the number Xr = 1234567890e-41. After normalization, this number will be equal to xr = 1.234567890e-032. You can insert these numbers into the program, or any other, to make sure.

vinc17fr, 2019-10-30:
The issue is the following. For instance, take N = 2 to round on 2 decimal digits, and say that you got a value ≈1.24999999999999 that needs to be rounded. The decimal result should be 1.2. But due to rounding errors in the program, you might have got ≈1.250000000001, whose rounding is 1.3, so that the program would return 1.3 instead of 1.2.

yur-spiridonov, 2019-10-30:
The number 1.25 is a representable number, therefore, when rounding to N = 2 with an error <= 0.05 it will give 1.25≈3. This is what my program gives.
You, apparently, meant another case. Insert the number x = 1.15 into the program. Then on the monitor we will see X = double (x) = 1.1499999999999999. Is that what you are talking about? My program gives the correct result: xr = 1.1

vinc17fr, 2019-10-30:
No, I'm not talking about 1.25, but the value that precedes it for instance. Anyway, there is the same issue with 1.15. Your program may give the correct result on it, but there might be other cases for which it doesn't. Actually I'm surprised that your program returns the correct answer on 1.15 (I can't test it because of bad formatting). Consider the following C program, which should do the same thing as yours on this particular example:
```
#include <stdio.h>

int main (void)
{
  double x = 1.15;
  double y = 10 * x;

  printf ("%.23g\n", x);
  printf ("%.23g\n", y);
  printf ("%lld\n", (long long int) (y + 0.5));
  return 0;
}
```

I get the following:
```
1.1499999999999999111822
11.5
12
```
Thus your program should have given 1.2 (which is an incorrect answer).

vinc17fr, 2019-10-30:
I've found the reason: your program actually does `(x * 100) / 10` instead of just `x * 10`. This gives the correct result, but just because you are lucky (I assume that there are other values for which this is the other way round, as you have two roundings instead of just one). Moreover, that's rather inefficient!

vinc17fr, 2019-10-30:
Indeed, take for instance N = 2 and X = 1.45; the exact double-precision value is below 1.45, but your program gives 1.5 instead of 1.4.

yur-spiridonov, 2019-10-30:
There is only one rounding in my program, this is Xr = static_cast <unsigned long long int> (X + 0.5); And two options for calculating X. Where X = x * pow (10, k) and X = X / 10. These two options cover all possible cases. Since k has a limited range of values, the function pow (10, k) can be specified in a table.

vinc17fr, 2019-10-30:
No, when you do a multiplication or a division, there is a rounding. With your program on 1.45 as a double, which is about 1.4499999999999999555911, there is first a multiplication by 100 since k = 2. This gives 145 (exactly). Then you have a division by 10, which gives 14.5 (exactly). Then you compute 14.5 + 0.5, which gives 15, converted to an integer, which is 15. Then it is divided by 10, yielding 1.5.

yur-spiridonov, 2019-10-30:
In my program, when rounding, I need to replace 0.5 with 0.499999999999999. Then everything will work correctly.

#include <iostream>
#include <math.h>
#include <stdlib.h>
#include <iomanip>
using namespace std;

int main()
{
double x,xr,X;
unsigned long long int Xr;
int N,E,e,k;

cout <<"Input a decimal precision N=";
cin>>N;

cout <<endl<<"Input a number and press ENTER:"<<"\n"<<"x= ";
cin>>x;
  //x=1.2345678901234e-32 ;

cout<<"X= "<< setprecision(18)<<x << '\n';

         frexp (x, &e);
E =static_cast <int> (e*0.301);
k=N-E;
if (E<0) //for format xr=d0.d1...dN10^E (d0≠0).
k=k+1;
  X=x*pow(10,k);
if (X > pow (10,N)){
X=X/10;
k=k-1;
}

   X=X+0.499999999999999;
   Xr=static_cast <unsigned long long int> (X);
   xr=Xr*pow(10,-k);

cout<<endl <<"Xr= "<<Xr<<"e"<<-k<<'\n';
cout<<"xr="<<xr<<'\n';

  return 0;
}

yur-spiridonov, 2019-10-30:
Remember to replace  X = xpow (10, k) with X = x * pow (10, k)

yur-spiridonov, 2019-10-30:
When multiplied by 100, we get ≈ 144.9999999999999999555911 = 1.4499999999999999555959ee2. We do not round this number. Then we divide it by 10 and get ≈14.49999999999999999555911. If this number is added to 0.499999999999999 and the whole part is discarded, then we get 1.4.
There can be no rounding to 1.45 in the processor, because This is a nonrepresentable number.

stroobandt, 2019-10-30:
As original poster of this issue almost two years back, here are some of my thoughts…

1. [**ANSI/IEEE Std 754-2019 did not deliver**](http://754r.ucbtest.org/background/) on my request to do something about this. This issue was directly referenced.
2. This was predicted by Paul Zimmermann of [LORIA](https://fr.wikipedia.org/wiki/Laboratoire_lorrain_de_recherche_en_informatique_et_ses_applications) in the ensuing discussion:
> instead of waiting 10 years that IEEE 754 standardizes correct rounding for
> binary to/from decimal conversions in all cases [1], and maybe another 10 years
> that it is implemented in your prefered language, my advice would be to use right
> now a language where it is implemented, for example the GNU MPFR library: […]
> 
> [1] Section 5.12.2 of IEEE 754-2008 says "There might be an implementation-defined
> limit on the number of significant digits that can be converted with correct
> rounding to and from supported binary formats."
3. The same Paul Zimmermann later followed up with another, more interesting suggestion to our dear Mr Spiridonov:
> Dear Iouri,
> you'll find in the book ["Modern Computer Arithmetic"](https://members.loria.fr/PZimmermann/mca/pub226.html), Algorithm 3.10,
> an algorithm for binary to decimal conversion which guarantees correct
> rounding.
4. It is pitiful that in this day and age ordinary developers (like myself) still have to deal with such trivial issues like this rounding issue.
5. Some members on the IEEE 754 commission gain their pay check way too easy indulging in academic discussions without solving the trivial real-world issues…

Only adding my 5 cents here …
I would be happy as a dog if only the built-in Micropython `round()` would do what most humans (including kids) would expect it to do, which is rounding correctly!

dpgeorge, 2019-10-31:
Note that part of the original bug reported at the very beginning, the part concerning printing/formatting of floats, was fixed a while ago in bc12eca461a317df842ce2e616afa97670cd0ce3 (thanks @jimmo for realising this).

So now running the OP test on PYBv1.1 gives:
```
MicroPython v1.11-533-g4e1b03d45 on 2019-10-31; PYBv1.1 with STM32F405RG

1011.640072 = float
1011.599898 = float rounded to one decimal
1011.6 = formatted rounded float

1011.600018 = float correctly rounded to least significant first decimal
1011.6 = formatted correctly rounded float

0.000119 = 1000 × machine epsilon
1.192093e-07
```

yur-spiridonov, 2019-10-31:
I think that the case when the error is 0.5ulp is a special case. This case must be recognized and processed in a special way. The following is a [program](https://onlinegdb.com/rJetmSc_qr) where this case is recognized.

The problem of processing a special case is that after the operation of multiplying the  x by 10 ^ k, we get the exact integer. This number is rounded to the nearest integer but we do not know from what number it is derived. I think that this information can be extracted from the expanded format of the product, before the number is rounded in the processor.
Consider two cases for the product of two double numbers in the extended format:
a) x=1.35
X=1.35000000000000008881784197001E0
1.0101100110011001100110011001100110011001100110011010
10^2 =1100100
X= X*10^2 = 1.0101100110011001100110011001100110011001100110011010*1100100=
10000111.000000000000000000000000000000000000000000000|011≈135
b) 1.45=1.44999999999999995559107901499E0 
1.0111001100110011001100110011001100110011001100110011
10^2 =1100100
1.0111001100110011001100110011001100110011001100110011*1100100=
10010000.1111111111111111111111111111111111111111111|111≈ 145
As we can see, in both cases, the processor returns us the exact integers. But in the first case, the protective bit = 0, and in the second case, the protective bit = 1. From this it follows that if we read the protective bit from the register and it = 0, then Xr = Xr + 5, if the protective bit = 1, then Xr = Xr-5

yur-spiridonov, 2019-11-01:
`` `         #include <iostream>
#include <math.h>
#include <stdlib.h>
#include <iomanip>
using namespace std;#include <iostream>`` `  
#include <math.h>
#include <stdlib.h>
#include <iomanip>
using namespace std;

yur-spiridonov, 2019-11-01:
Here is a variant of the [program in the online compiler](https://onlinegdb.com/BkywNKY5r) , which takes into account special cases.

yur-spiridonov, 2019-11-02:
The number 65.506 is correctly rounded to 65.51. But 65.51 is not a representable number. Therefore, in a double, this number is represented as 6.55100000000000051159076974727E1 = 6.551000000000001

yur-spiridonov, 2019-11-02:
Mr. Lefèvre  pointed out cases where rounding by the algorithm presented above gives an incorrect result. For example, the number 1.45 will be represented in the internal double format as 1.44999999999999995559107901499E0≈1.4. When rounding to N = 2, after multiplying by 100, in the internal representation we get the exact integer 145. Dividing it by 10 and adding it to 0.5, we get the rounded number 15 * 10 ^ -1, although we should get 1.4. The following is a program that takes these special cases into account. You can get more information[ here](https://github.com/yur-spiridonov/Decimal-rounding-of-binary-numbers/commit/2084c8aeb28a785dee7fc98822b575da09194bbd).

```#include <iostream>
#include <math.h>
#include <stdlib.h>
#include <iomanip>
using namespace std;

int main()
{
   double InputNumber,RoundNumber,X,TruncatedNumber,TruncateX,NormTruncatedNumber,Sub;
   unsigned long long int WholeRoundNumber;
   int DecimalPrecisionN,DecimalExponent,BinaryExponent,k;
   int SignSpecialCase = 0;
  cout <<"Input a decimal precision N=";
  cin>>DecimalPrecisionN;
  cout <<endl<<"Input a number and press ENTER:"<<"\n"<<"x= ";
  cin>>InputNumber;
   cout<<"X= "<< setprecision(20)<<InputNumber << '\n';

      frexp (InputNumber, &BinaryExponent);
    DecimalExponent =static_cast <int> (BinaryExponent*0.301);
    k=DecimalPrecisionN-DecimalExponent;
   if (DecimalExponent<0)
        k=k+1;
    X=InputNumber*pow(10,k);

       if (X > pow(10,DecimalPrecisionN)){
            X=X/10;
            k=k-1;
      }
/////////////////////////////////////////////////
//Definition of a special case attribute
    TruncateX=static_cast<int>(X);
    if (X-TruncateX==0.5)
      SignSpecialCase=1;

   TruncatedNumber=static_cast <unsigned long long int> (InputNumber*pow(10,14));
   NormTruncatedNumber=TruncatedNumber*pow(10,-14);
   Sub= NormTruncatedNumber-InputNumber;

   if (Sub==0)
   SignSpecialCase=0;

//////////////////////////////////////////////////////////
       WholeRoundNumber=static_cast <unsigned long long int> (X+0.5);

        if (SignSpecialCase==1)
             WholeRoundNumber=WholeRoundNumber-1;
        RoundNumber=WholeRoundNumber*pow(10,-k);

    cout<<endl <<"Rounded integer mantissa= "<<WholeRoundNumber<<"e"<<-k<<'\n';
    cout<<"Round number="<<RoundNumber<<'\n';

   system("pause");

      return 0;
}

yur-spiridonov, 2019-11-03:
There are decimal numbers that can be attributed to a special case. These are numbers that have N 1 significant digits and end with 5. For example, for rounding to N = 2, these are numbers 1.35,1.45, and for rounding to N = 7 it is 1.2345675, etc. So, the number 1.35 has an internal double representation equal to 1.35000000000000008881784197001E0 and therefore for N = 2 it should be rounded to 1.4. And the number 1.45 has an internal representation of 1.44999999999999995559107901499E0 and should be rounded to 1.4. For proper rounding, it is necessary to identify such cases and take them into account when rounding. The following is a program that takes these special cases into account.

```#include <iostream>
#include <math.h>
#include <stdlib.h>
#include <iomanip>
using namespace std;

int main()
{
   double InputNumber,RoundNumber,X,TruncatedNumber,TruncateX,NormTruncatedNumber,Sub;
   unsigned long long int WholeRoundNumber;
   int DecimalPrecisionN,DecimalExponent,BinaryExponent,k;
   int SignSpecialCase = 0;
  cout <<"Input a decimal precision N=";
  cin>>DecimalPrecisionN;
  cout <<endl<<"Input a number and press ENTER:"<<"\n"<<"x= ";
  cin>>InputNumber;
   cout<<"X= "<< setprecision(20)<<InputNumber << '\n';

      frexp (InputNumber, &BinaryExponent);
    DecimalExponent =static_cast <int> (BinaryExponent*0.301);
    k=DecimalPrecisionN-DecimalExponent;
   if (DecimalExponent<0)
        k=k+1;
    X=InputNumber*pow(10,k);

       if (X > pow(10,DecimalPrecisionN)){
            X=X/10;
            k=k-1;
      }
/////////////////////////////////////////////////
//Definition of a special case attribute
    TruncateX=static_cast<int>(X);
    if (X-TruncateX==0.5)
      SignSpecialCase=1;

   TruncatedNumber=static_cast <unsigned long long int> (InputNumber*pow(10,14));
   NormTruncatedNumber=TruncatedNumber*pow(10,-14);
   Sub= NormTruncatedNumber-InputNumber;

   if (Sub==0)
   SignSpecialCase=0;

//////////////////////////////////////////////////////////
       WholeRoundNumber=static_cast <unsigned long long int> (X+0.5);

        if (SignSpecialCase==1)
             WholeRoundNumber=WholeRoundNumber-1;
        RoundNumber=WholeRoundNumber*pow(10,-k);

    cout<<endl <<"Rounded integer mantissa= "<<WholeRoundNumber<<"e"<<-k<<'\n';
    cout<<"Round number="<<RoundNumber<<'\n';

       return 0;
}

neilpanchal, 2019-11-03:
@yur-spiridonov We already have [MPFR](https://www.mpfr.org/) and many other [C libraries](https://en.wikipedia.org/wiki/List_of_arbitrary-precision_arithmetic_software). Why implement a yet another C routine?

stroobandt, 2020-02-21:
> @yur-spiridonov We already have [MPFR](https://www.mpfr.org/) and many other [C libraries](https://en.wikipedia.org/wiki/List_of_arbitrary-precision_arithmetic_software). Why implement a yet another C routine?

The original issue with examples can be found [here](https://github.com/micropython/micropython/issues/3516#issue-284453215).

govirginia, 2020-05-05:
In Excel, 
  round(1011.640071,1)=1011.6000000000000000000000000000
  round(1011.640071,2)=1011.6400000000000000000000000000
Excel has had this correct for decades, so why so much trouble to copy its methods to Python or microPython?  Dont have to re-invent the wheel.

yur-spiridonov, 2020-05-07:
Calculations in Excel are performed in double in accordance with the specification of the IEEE754 standard. The result is displayed in decimal string. Rounding of the output in Excel is performed on decimal according to the rules of decimal arithmetic. The problem with Python and other programming languages that use floating point numbers is to find the binary number closest to a properly rounded decimal without going to decimal numbers. Why this is needed can be demonstrated by the following example in Excel.
Let A1 = 1.234567890987, A2 = 1.23456789, A3 = A1 - A2 = 9.87E-10,
A4 = A3 * 10 ^ 5 - 9.87 * 10 ^ -5
In decimal, we should get the answer - 0.
Excel returns the number 3.7255825188113900E-12
The binary equivalent of this number is stored in the internal double, and is involved in further calculations.
But if we calculate A5 = round (A3,15), then we get A4 = A5 * 10 ^ 5 - 9.8 * 10 ^ -4 = 0.
Note that A5 = 9.87E-10 and a coefficient of 9.87 * 10 ^ -5 are non-representable numbers. However, the difference A4 is 0.
This leads to the conclusion that for further correct binary calculations, the double (A1 - A2) should be reduced to double A5, which is equal to the correctly rounded decimal number round (A3,15).
An algorithm that allows you to find such a number without converting to a decimal string is called the correct rounding of decimal numbers in binary code.

vinc17fr, 2020-05-07:
> Calculations in Excel are performed in double in accordance with the specification of the IEEE754 standard.

I don't know whether this has changed, but for a very long time, this has not been true: it had wrappers to try avoiding the effect of cancellation. OpenOffice even copied it, which annoyed (some) users (there was a long thread in the French mailing-list in 2005).

> The result is displayed in decimal string. Rounding of the output in Excel is performed on decimal according to the rules of decimal arithmetic.

Decimal arithmetic is about decimal. Here you have a conversion from binary to decimal, and there exist different rules.

> The problem with Python and other programming languages that use floating point numbers is to find the binary number closest to a properly rounded decimal without going to decimal numbers.

This is different here in Python, because the goal of the `round()` function is to get a rounding in decimal on some given number of digits, but keeping the representation in binary. With a naive implementation, you would do a conversion from binary to decimal (like with a usual conversion into a string), then another conversion from decimal to binary.

> Why this is needed can be demonstrated by the following example in Excel.
> Let A1 = 1.234567890987, A2 = 1.23456789, A3 = A1 - A2 = 9.87E-10,
> A4 = A3 * 10 ^ 5 - 9.87 * 10 ^ -5
> In decimal, we should get the answer - 0.
> Excel returns the number 3.7255825188113900E-12

Note that in your example, you have a rounding for A1 and A2 in the decimal-to-binary conversion. A3 is computed exactly (Sterbenz's lemma). You have again several roundings in the computation of A4.

> The binary equivalent of this number is stored in the internal double, and is involved in further calculations.
> But if we calculate A5 = round (A3,15), then we get A4 = A5 * 10 ^ 5 - 9.8 * 10 ^ -4 = [...]

What you say does not make sense. You have ignored many roundings.

mdickinson, 2020-05-07:
Maybe there's a user-education opportunity here? You could have two-argument `round` raise `NotImplementedError("Sorry, this is not implemented for MicroPython. You probably want string formatting instead.")`

I still dream of one day deprecating the two-argument form of `round` in Python. Unfortunately, I don't think it's a realistic dream: there would be too much gratuitous breakage of existing code.

mdickinson, 2020-05-07:
> With a naive implementation, you would do a conversion from binary to decimal (like with a usual conversion into a string), then another conversion from decimal to binary.

For what it's worth, that's *exactly* what CPython does, at least on most machines, using David Gay's well-known dtoa work. [Here](https://github.com/python/cpython/blob/8963a7f1f84a05412178b56629508b660d38861b/Objects/floatobject.c#L937) is the (correctly-rounded) conversion from binary float to decimal string, and [here](https://github.com/python/cpython/blob/8963a7f1f84a05412178b56629508b660d38861b/Objects/floatobject.c#L962) is the (correctly-rounded) conversion back from decimal to binary.

yur-spiridonov, 2020-05-07:
> This is different here in Python, because the goal of the `round()` function is to get a rounding in decimal on some given number of digits, but keeping the representation in binary.

If so, then we are only talking about formatting the decimal string for ease of perception and has no mathematical meaning.

> With a naive implementation, you would do a conversion from binary to decimal (like with a usual conversion into a string), then another conversion from decimal to binary.

It seems to me that this makes sense if you perform some actions on decimal numbers, for example, round them correctly. 

> A3 is computed exactly (Sterbenz's lemma). You have again several roundings in the computation of A4.

The exact value of the number A3, calculated manually, is 9.87E-10. The decimal equivalent of this number converted to double (after binary rounding to 53 significant digits) will be 9.87000000000000003776083043255E-10. At the same time, Excel, like Python, and C ++ will return the number 9.8700003725582500E-10, which is not the closest to the exact value of A3.

> You have again several roundings in the computation of A4.

You're right. But these are binary roundings. If in our example we operated with binary codes of correctly rounded decimal numbers, we would get the binary number double, closest to the exact value of 9.87E-10.

> What you say does not make sense. You have ignored many roundings.

You are right, my last statement is wrong. It is true in this edition. The difference A4 = A5 * 10 ^ 5 - 9.8 * 10 ^ -4 will be exactly 0 if both terms represented in double are correctly rounded. In our case, this means that the binary number A5 * 10 ^ 5 will be reduced to such a binary number that will be the closest to a properly rounded decimal number. We have this number 9.8 * 10 ^ -4.

yur-spiridonov, 2020-05-07:
> Here is the (correctly-rounded) conversion from binary float to decimal string,

Check it [out]([url](https://github.com/yur-spiridonov/Decimal-rounding-of-binary-numbers/commit/2084c8aeb28a785dee7fc98822b575da09194bbd))