micropython/micropython #3517: .seek(offset, SEEK_CUR) not supported

dhalbert, 2017-12-26:
Seeking to a position to a file relative to the current offset is not supported.`somefile.seek(offset, SEEK_CUR)` returns `MP_ENOTSUPP`.

https://github.com/micropython/micropython/blob/5de064fbd056944e393be9355855eee6562fecdb/extmod/vfs_fat_file.c#L136

Since `f_tell()` is implemented (and used in that routine), it seems to me the code could call `f_tell()` and then call `f_lseek()` adding the offset to `f_tell()` value. Is there some reason I'm missing that it's not supported?

(Noticed by a CircuitPython user: https://github.com/adafruit/circuitpython/issues/501).

dhalbert, 2018-01-09:
And see this PR, which implements `SEEK_CUR`:  https://github.com/micropython/micropython/pull/3494

dpgeorge, 2018-01-31:
This is now fixed by 7642785881550bd71815623c4f93e0516217429c

closed 2018-01-31 by dpgeorge