micropython/micropython #3518: How to connect to an ESP32 from a mobile application?

ShrikantKumbhar, 2017-12-26:
I want to communicate ESP32 with my mobile app. I done micopython coding as follows :

import network
ap = network.WLAN(network.AP_IF)
ap.active(True)
ap.config(essid='ESP32')
ap.config(authmode=3, password='123456789')

In this code Access point is created and it is visible in my android mobile. 
But when i connect my mobile to access point it is showing “saved” message instead of “connected” even though i have entered correct password. 
I followed same procedure using arduino code then i am able to connect my mobile with access point created by esp32. Any help is highly appreciable .

robert-hh, 2017-12-26:
I have no issues with your script example. I can connect with an android tablet and exchange data.

dpgeorge, 2019-06-24:
AP mode on the esp32 should be working, so I'll close this issue.

closed 2019-06-24 by dpgeorge