micropython/micropython #3534: Micropython -m upip command not working

nikhil1983, 2018-01-02:
Hey,
I am currently working on stm32f4disc board. I am installing uasyncio library using the following commands. 

> micropython -m upip install -p ~/syn micropython-uasyncio
> micropython -m upip install -p ~/syn micropython-uasyncio.queues
> micropython -m upip install -p ~/syn micropython-uasyncio.synchro

So whenever i input first command, it gives me an output of micropython command not found 

> edutech@edutech-desktop:~$ micropython - m upip install micropython-binascii
> micropython: command not found
> edutech@edutech-desktop:~$ 

So can anyone help me in installing uasyncio ?

robert-hh, 2018-01-02:
In the README.md of micropython lib you can read:

> **Following examples assume that `micropython` binary is available on your `PATH`:**
> 
> ~~~~
> $ micropython -m upip install micropython-pystone
> 

These commands however install usyncio on your computer, not on the stm32f4disc board. But it creates the file structure, you either can copy to your board or into the modules directory of your build.

nikhil1983, 2018-01-02:
Is there any document of how to add micropython binart in your PATH ? 
I have downloaded the micropython-lib from github, unzip it and it has many folders name uasyncio. Which one should i copy on my target board ?

robert-hh, 2018-01-02:
To be included into the frozen bytecode, you can copy the files manually to your <board>/modules directory. Create there the subdirectory uasyncio, and copy into it:
__init__.py from https://github.com/micropython/micropyt ... o/uasyncio
core.py from https://github.com/micropython/micropyt ... e/uasyncio
queues.py from https://github.com/micropython/micropyt ... s/uasyncio
synchro.py from https://github.com/micropython/micropyt ... o/uasyncio
Or you create the subdir uasyncio in the flash filesystem of your board and copy these files into it.

nikhil1983, 2018-01-02:
Okay.I created subdir in flash Filesystem of my board and its working now. **Thanks for help.**

nikhil1983, 2018-01-02:
Can you tell me how to set micropython binary is set in PATH. Because if i want to use this command path should be set.

> micropython -m upip install -p ~/syn micropython-uasyncio   

At present it will give me error of **micropython command not found** as PATH isn't set.
Is there any document available for setting micropython binary to PATH or just you can tell the steps, i will implement it. 

**Thanks** again for help :)

robert-hh, 2018-01-02:
Good. may I remind you to close the issue then.

closed 2018-01-02 by nikhil1983