micropython/micropython #3537: ESP32: HelTec's Wifi Kit 32 fails to connect to AP

Lisa999, 2018-01-02:
Using the latest standard  build (after make erase):

lisa@RackServer:esp32$ esptool.py --port /dev/ttyUSB3 write_flash --compress 0x1000 esp32-20180102-v1.9.3-238-g42c4dd09.bin 
esptool.py v2.1
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 935888 bytes to 587150...
Wrote 935888 bytes (587150 compressed) at 0x00001000 in 52.3 seconds (effective 143.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
lisa@RackServer:esp32$ picocom -b 115200 /dev/ttyUSB3 | tee ~/esp32-logfile
Terminal ready

>>> 
>>> import network as wifi
>>> nic = wifi.WLAN(wifi.STA_IF)
I (59634) wifi: wifi firmware version: 111e74d
I (59634) wifi: config NVS flash: enabled
I (59634) wifi: config nano formating: disabled
I (59634) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (59644) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (59664) wifi: Init dynamic tx buffer num: 32
I (59664) wifi: Init data frame dynamic rx buffer num: 64
I (59664) wifi: Init management frame dynamic rx buffer num: 64
I (59664) wifi: wifi driver task: 3ffe1584, prio:23, stack:4096
I (59674) wifi: Init static rx buffer num: 10
I (59674) wifi: Init dynamic rx buffer num: 0
I (59674) wifi: Init rx ampdu len mblock:7
I (59684) wifi: Init lldesc rx ampdu entry mblock:4
I (59684) wifi: wifi power manager task: 0x3ffe84b0 prio: 21 stack: 2560
W (59694) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (59894) phy: phy_version: 362.0, 61e8d92, Sep  8 2017, 18:48:11, 1, 2
I (59904) wifi: mode : null
>>> nic.active(True)
I (59904) wifi: mode : sta (30:ae:a4:4c:52:10)
I (59914) wifi: STA_START
True
>>> nic.connect('xxx', 'xxx')
>>> I (79554) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (80114) wifi: state: init -> auth (b0)
I (81114) wifi: state: auth -> init (2)
I (81114) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (81114) wifi: STA_DISCONNECTED, reason:2
I (81244) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (81244) wifi: state: init -> auth (b0)
I (82244) wifi: state: auth -> init (2)
I (82244) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (82244) wifi: STA_DISCONNECTED, reason:2
>>> nic.disconnect()
I (93794) wifi: state: auth -> init (0)
I (93794) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (93794) wifi: STA_DISCONNECTED, reason:202
authentication failed
>>> 
Thanks for using picocom
picocom v1.7

port is        : /dev/ttyUSB3
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

robert-hh, 2018-01-03:
So you're sure you used the right SSID and password in nic.connect()?

Lisa999, 2018-01-03:
@robert-hh: ssid and password are correct. Using open wifi network makes no difference (connect using nic.connect('xxx-guests','') )
Board has built-in antenna: https://ae01.alicdn.com/kf/HTB1XI91SpXXXXX5aVXXq6xXFXXX0/Internet-Development-Board-ESP32-WIFI-chip-0-96-inch-OLED-Bluetooth-WIFI-Kit-32-module-for.jpg
Verified it by looking at the bottom, and yes the picture is identical.
I'm lost and stumped, have no clue why it ain't connecting regardless ssid (used different wifi AP, same result)

dpgeorge, 2018-05-23:
Please try again with the latest ESP32 firmware.

Agusss19, 2018-06-07:
I have the same problem with the same ESP32 chip

dpgeorge, 2018-06-08:
> wifi: STA_DISCONNECTED, reason:2

This corresponds to WIFI_REASON_AUTH_EXPIRE.  You might want to try completely erasing the flash then putting the firmware back on it.

If that doesn't work then it could be a hardware problem, see eg https://esp32.com/viewtopic.php?t=1072

Lisa999, 2019-03-26:
Old, time to close.

closed 2019-03-26 by Lisa999

HerrSpace, 2019-05-30:
I'm seeing this too, just tested it on a second board to be sure. Pretty much every second attempt fails.  I'm using the example do_connect function. ```I (442) cpu_start: Pro cpu up.
I (442) cpu_start: Application information:
I (442) cpu_start: Compile time:     07:44:16
I (444) cpu_start: Compile date:     May 29 2019
I (450) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (456) cpu_start: Starting app cpu, entry point is 0x4008294c
I (0) cpu_start: App cpu up.
I (467) heap_init: Initializing. RAM available for dynamic allocation:
I (473) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (479) heap_init: At 3FFB9B88 len 00026478 (153 KiB): DRAM
I (486) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (492) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (498) heap_init: At 40093398 len 0000CC68 (51 KiB): IRAM
I (505) cpu_start: Pro cpu start user code
I (75) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (100) modsocket: Initializing
I (160) wifi: wifi driver task: 3ffe2f68, prio:23, stack:3584, core=0
I (397) wifi: wifi firmware version: 38e2484
I (397) wifi: config NVS flash: enabled
I (397) wifi: config nano formating: disabled
I (397) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (407) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (447) wifi: Init dynamic tx buffer num: 32
I (447) wifi: Init data frame dynamic rx buffer num: 32
I (447) wifi: Init management frame dynamic rx buffer num: 32
I (447) wifi: Init static rx buffer size: 1600
I (447) wifi: Init static rx buffer num: 10
I (457) wifi: Init dynamic rx buffer num: 32
I (527) phy: phy_version: 4007, 9c6b43b, Jan 11 2019, 16:45:07, 0, 0
I (527) wifi: mode : sta (3c:71:bf:8c:07:54)
I (527) wifi: STA_START
connecting to network...
I (1867) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2437) wifi: state: init -> auth (b0)
I (2447) wifi: state: auth -> init (8a0)
I (2447) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2447) wifi: STA_DISCONNECTED, reason:202
authentication failed```

dpgeorge, 2019-05-31:
@HerrSpace are you using a HelTec? did you double check the SSID and password?

To make progress on the issue you would need to compile a standard ESP IDF example that connects to a WiFi access point and see if that has the same error or not (to see if the issue is with the hardware/IDF, or with MicroPython).

HerrSpace, 2019-06-02:
Ah, I missed this issue being about HelTec. I do not have a HelTec board, but this pretty generic dev board: https://www.az-delivery.de/products/esp32-developmentboard
It might be a different revision though.

The SSID and password are correct. As I said, it only fails in like 50% of the reboots.

I'm testing the wifi sta example in esp idf now.

HerrSpace, 2019-06-02:
Okay, here we go:

I flashed the wifi station example from the esp idf. There are a couple of 'retry to connect to the AP' lines in there, but it always eventually manage to connect to my wifi:
https://gist.github.com/HerrSpace/976511128d4d68880c82b5fed92b6919

Then I reflashed micropython and upload the following snippet as main.py 

```python
import network

SSID = '<myssid>'
PASSWORD = '<mypass>'

wlan = network.WLAN(network.STA_IF)
wlan.active(True)
if not wlan.isconnected():
    print('connecting to network...')
    wlan.connect(SSID, PASSWORD)
    while not wlan.isconnected():
        pass
print('network config:', wlan.ifconfig())
```

Which results in every second reboot failing to connect to the AP:
https://gist.github.com/HerrSpace/967c7dbff36e01b1bc2843cff2cecae7

dpgeorge, 2019-06-02:
Thanks for the logs.  The difference between the IDF example and your MicroPython code is that the former will automatically try to reconnect, regardless of what error may have caused the connection attempt to fail, while the latter will only try to reconnect if there was a beacon timeout or the AP was not found (but not in the case of a failed authentication because likely that means the password was wrong).

In MicroPython it's not a good idea to have a simple `while not wlan.isconnected()` loop, because the connection may fail, in which case the code will need to handle that failure.  Eg:
```python
while not wlan.isconnected():
    print('attempting connection...')
    wlan.connect(...)
    while wlan.status() == wlan.STAT_CONNECTING:
        pass
```

HerrSpace, 2019-06-03:
Why not adopt the esp idf behavior of retrying on authentication failure? I get your point, but the esp folks seem to disagree with the conclusion that we shouldn't try reconnecting. My wifi setup isn't homegrown either. I's a very common home router (FRITZ!Box 6490 Cable).  

Regarding your wifi connection code: Maybe consider changing the esp32 reference code for wifi connections then. That's where my code was from:
https://docs.micropython.org/en/latest/esp32/quickref.html?highlight=wifi#networking

dpgeorge, 2019-06-04:
> Why not adopt the esp idf behavior of retrying on authentication failure?

Yes that's probably a good idea.  Reading the docs, it seems that `WIFI_REASON_AUTH_FAIL` doesn't necessarily mean the password is wrong, and a wrong password may also lead to a different reason.  And I guess with IoT devices (where you own the device and the network) it's much more likely that the connection failed due to a temporary failure rather than a wrong password, so making it retry seems like the right thing to do.

@HerrSpace do you have an esp8266 which you could test wifi connection with?  It'd be interesting to see how that behaves, what messages it prints (if you try, please use a latest development build, not a release build, so that debug messages are enabled).

dpgeorge, 2019-06-04:
> and a wrong password may also lead to a different reason.

Indeed: using an incorrect password I get `WIFI_REASON_AUTH_EXPIRE` and then `WIFI_REASON_NO_AP_FOUND`

HerrSpace, 2019-06-04:
> Yes that's probably a good idea. 

Cool!

> do you have an esp8266 which you could test wifi connection with?

Errr, I have some but they are a pain to program. I'll try to make them work tonight.

HerrSpace, 2019-06-08:
Wow I hate these chips, BUT: I can't reproduce this problem on my esp8266 e12 board. Using the same main.py as above I consistently get:

```
connecting to network...
network config: ('192.168.178.86', '255.255.255.0', '192.168.178.1', '192.168.178.1')

MicroPython v1.11-8-g48dcbbe60 on 2019-05-29; ESP module with ESP8266
Type "help()" for more information.
>>>
```

HerrSpace, 2019-06-08:
^ An MR to include your loop into the example do_connect.

HerrSpace, 2019-06-11:
Joo @dpgeorge should I create a new issue for this? Also can you have a look at my MR?

dpgeorge, 2019-06-12:
> BUT: I can't reproduce this problem on my esp8266 e12 board. Using the same main.py as above I consistently get:

Ok, thanks for testing.  So I think it'd be good to make the esp32 behave the same way and just always try to reconnect regardless of the error it gets when attempting to connect.  In #4842 I've made such a change.