micropython/micropython #3542: PWM - set duty with negative number leads to full PWM

EddieParis, 2018-01-04:
This is because the function pwm_set_duty() in file esppwm.c has uint16_t as argument. Hence if in python we pass a negative number the function receives a huge positive number.

Solution would be to change uint16_t to int16_t and add a warning about that in the doc.

This change would save me some line of code when I decrease the pwm for my dimmer :-)

robert-hh, 2018-01-04:
Instead e.g.
`duty -= step`
you could use
`duty = max(1, duty - step)`

EddieParis, 2018-01-04:
Yes of course, there is a lot of way to avoid this in python, but my point is it shall be done in the C code directly. This would avoid some weird surprises and save some useless check on python side !

I is already clipped correctly for values above 1023, so consistency would be to do it correctly for values below 0.

EddieParis, 2018-02-13:
Created pull request #3609

pmp-p, 2018-02-13:
Thank you now i understand some gliches :D , indeed clipping should be both ends.

EddieParis, 2018-02-22:
@pmp-p I found another glitch source, I wanted you to be aware of it.
When you have only 1 PWM channel active if you set it to 1023 then to any other value (even close like 1000) there is a black flash. I found the root cause,  I'm preparing a new ticketa and patch for it.

dpgeorge, 2018-03-07:
This was fixed in 5c83d05b493af58bb14cba9dbfc2bf2d85de2962

closed 2018-03-07 by dpgeorge