micropython/micropython #3548: Built-in functon next() should be able to take a second optional argument.

stroobandt, 2018-01-04:
Quoting https://docs.python.org/3.5/library/functions.html#next

> 
> **next**(_iterator_[, _default_])
> 
> Retrieve the next item from the _iterator_ by calling its [`__next__()`](https://docs.python.org/3.5/library/stdtypes.html#iterator.__next__) method. If _default_ is given, it is returned if the iterator is exhausted, otherwise [`StopIteration`](https://docs.python.org/3.5/library/exceptions.html#StopIteration) is raised.

dpgeorge, 2018-01-31:
This incompatibility is actually documented here: http://docs.micropython.org/en/latest/pyboard/genrst/modules.html#second-argument-to-next-is-not-implemented

Use of a second argument to next() doesn't seem to be that common, and since a work-around exists (see doc link above) it seems best not to add this capability to uPy to keep it minimal.

stroobandt, 2018-01-31:
Indeed, Damian, I found the documented work-around only shortly after posting this issue. It works fine.
However, I decided not to delete this issue to show that, yes, there is some interest in `next()`.
Actually, I have code where I am using two in a row.
Anyhow, it is Python 3.5; not 3.4.

closed 2018-01-31 by stroobandt

dpgeorge, 2018-01-31:
> However, I decided not to delete this issue to show that, yes, there is some interest in next().

Yes, it's good to hear that, when there is interest in further refinements of compatibility with CPython.  If others also have interest then they can comment here.

(Note that adding a second arg to next() is actually quite simple to do.)