micropython/micropython #3552: Strategy for compiling big module .

curlyz, 2018-01-07:
[ INTRA-PACKAGE REFERENCE ]
Hi , I am using uPython on ESP32 chipset , interfacing with APDS9960.
There are working module for this chip online but it is for RPi . But easily ported to ESP32.
The problem is that the file is hugh and the ESP doesnt have enough RAM to compile it succesfully . 
So my strategy is to divide the module file. I found out that python have Intra-Package reference. I am wondering if anyone ever be in my situation ?
Thanks.
[apds9960 Module](https://github.com/liske/python-apds9960)

robert-hh, 2018-01-07:
If the source file is too large, you can either pre-compile irt with mpy-cross or embed it into the firmware as frozen bytecode. There is a lot of discussion and hint about both methods in the forum forum.micropython.org

curlyz, 2018-01-07:
Ok, I will look for frozen byte code. Thanks Robert

closed 2018-01-07 by curlyz