micropython/micropython #3555: ESP8266 I2C driver SCL glitch

ChrisMicro, 2018-01-07:
Compared to the Arduino I2C driver the driver for the ESP8266 seems to pull the SCL clock line for one or to machine cycles actively high. This causes problems for some I2C devices like the MPU6050.

curlyz, 2018-01-08:
please provide your code so that we can replicate the bug.

ChrisMicro, 2018-01-08:
I use this driver and example from here:
https://github.com/adamjezek98/MPU6050-ESP8266-MicroPython
There is a comment in the driver which gives a hint to the problem:
"def val_test(self): # ONLY FOR TESTING! Also, fast reading sometimes crashes IIC"
You can see the difference off the signals between the Micropython and the Arduino driver on the the ESP8266 and a full analysis of the problem here:
https://www.mikrocontroller.net/topic/442599#5269272

dpgeorge, 2018-01-31:
It looks like that MPU6050 driver is issuing `i2c.start()` and `i2c.stop()` commands when it shouldn't.  Eg it does:
```python
self.iic.start()
self.iic.writeto(self.addr, bytearray([107, 0]))
self.iic.stop()
```
when it should do simply: `self.iic.writeto(self.addr, bytearray([107, 0]))`.

@ChrisMicro I suggest you try that fix in the MPU6050 driver and if it works ask the maintainer of that driver to fix it.

peterhinch, 2018-01-31:
There is [this driver](https://github.com/micropython-IMU/micropython-mpu9x50) which supports the MPU6050.

ChrisMicro, 2018-01-31:
> @ChrisMicro I suggest you try that fix in the MPU6050 driver and if it works ask the maintainer of that driver to fix it.

In my opinion the problem is deeper in the I2C-driver for the ESP8266.
Let's look at the observations:
1. with the ESP-Python I2C driver I can not even make a I2C port scan
2. with the ESP-Arduino driver I can
3. with the ESP-Python I2C driver and a 12pF capacitor in the SCL line the port scan works

Therefore a closer look on the signals is required.
1. The ESP-Arduino driver works correct and the edge of the SCL-signal looks like this:
https://www.mikrocontroller.net/attachment/preview/352764.jpg
2. The ESP-Python I2C driver does not work correctly and looks like this:
https://www.mikrocontroller.net/attachment/preview/352763.jpg

Conclusion:
There is a difference between the ESP-Arduino-SCL signal and the ESP Python driver regarding the PIN-initialization. Probably it is not properly initialized as Open-Drain.
The wrong Signal of the ESP-Python-I2C driver leads to a problem for the I2C chips with certain SCL switching levels. Some chips may work others not.

durin42, 2018-01-31:
I'm seeing the same behavior as @ChrisMicro - with the exact same breadboarded circuit, I can't get an i2c scan to find anything with micropython, but an arduino sketch on the same board can scan and find the same device. I haven't had a chance to diagnose further than that - I'll see if I can get my hands on a 12pF cap soonish to test that.

peterhinch, 2018-01-31:
I don't have an MPU6050 but I successfully connected to an MPU9250 using the driver I cited above. This was on a WeMos D1 Mini using soft I2C. The MPU9250 is effectively a superset of the MPU6050.

ChrisMicro, 2018-01-31:
> I don't have an MPU6050 but I successfully connected to an MPU9250 

That might be possible. Either there is a small difference in the level detection of the MPUs in the SCL line or it might be that even chips of the same type may have different  behavior. The influence of a 12pF capacity might be really small. But at least I can see a difference in the oscilloscope graph and it seems to me that the Python driver pulls the SCL actively high which might lead to the malfunction.

durin42, 2018-01-31:
For what it's worth my current test code is just:

```
from machine import Pin, I2C
i2c = I2C(scl=Pin(5), sda=Pin(4), freq=5000)
i2c.scan()
```

at the REPL. In case that helps. Right now I get the empty list back, I don't have a tiny cap handy to test but I'll see if I can pick one up soon to check for a difference in behavior.

ChrisMicro, 2018-02-01:
> @durin42 I don't have a tiny cap handy to test but I'll see if I can pick one up soon to check for a difference in behavior.
If you could test it, that could be really helpful. I had a 12pF capacitor in my hands, so I took this one. It might be working with higher values. ( 1nF is to high, the edges are to flat )

dpgeorge, 2018-02-02:
> Therefore a closer look on the signals is required.

I had a more detailed look at the signals coming from I2C and I can confirm that there is an issue.  I can reproduce the images of the traces linked from mikrocontroller.net.

My test code looks like this:
```python
import machine
scl = machine.Pin(5, machine.Pin.OPEN_DRAIN, machine.Pin.PULL_UP)
sda = machine.Pin(4, machine.Pin.OPEN_DRAIN, machine.Pin.PULL_UP)
i2c = machine.I2C(scl=scl, sda=sda)
i2c.scan()
```
Note that I'm using the internal pull ups on the pins which may affect the output but I don't think that will change my findings.

On a scope (sorry no pics) I see that when SCL rises it has a sharp rise (assumed from driving the pin actively high) followed by a gradual rise (assumed from letting the pin float high).  This does seem like the wrong behaviour for open drain.

But, running the same test with SCL=Pin(12) and SDA=Pin(13) I don't get this strange behaviour.  Instead, SCL rises gradually the whole way.

So perhaps different pins behave differently in open drain mode.  (Note: pins 4 and 5 on my board seem to act strange because in push-pull output mode they are unable to drive the pin high.... so there might be a hardware issue here with my particular chip.)

Anyway, I tried to fix the problem on pins 4 and 5 by modifying the way the pin is driven high in open drain mode.  I changed:
```diff
--- a/ports/esp8266/esp_mphal.h
+++ b/ports/esp8266/esp_mphal.h
@@ -86,7 +86,7 @@ void mp_hal_pin_open_drain(mp_hal_pin_obj_t pin);
     } while (0)
 #define mp_hal_pin_od_high(p) do { \
         if ((p) == 16) { WRITE_PERI_REG(RTC_GPIO_ENABLE, (READ_PERI_REG(RTC_GPIO_ENABLE) & ~1)); } \
-        else { gpio_output_set(1 << (p), 0, 1 << (p), 0); } \
+        else { gpio_output_set(0, 0, 0, 1 << (p)); } \
     } while (0)
 #define mp_hal_pin_read(p) pin_get(p)
 #define mp_hal_pin_write(p, v) pin_set((p), (v))
```
And with this patch it seemed to fix things, the SCL line didn't have a sharp rise to high.

@ChrisMicro @durin42 can either/both of you please try two independent things: 1) use different pins, eg 12 and 13 and re-run your test with the MPU6050; 2) apply the 1-line patch above and re-run with pins 5 and 4.

chrismas9, 2018-02-02:
I did a documentation and code review and came to the same conclusion that switching the pin from output to input would be better. This is also what machine.pin does. There should be no need to set the pins to open drain mode as they are being used as output for low and input for high.

I can't find a published schematic of the GPIO block, but there are two ways Espressif could have implemented hardware open drain. 1) disable the top p-channel MOSFET in open drain mode to create a true open drain n-channel driver, or 2) multiplex the output enable to data in open drain mode. Based on the waveform pictures I would say they used the latter method. The data logic 1 propagates directly to the output driver, but is slightly delayed to the output enable by the multiplexer. The pin will start to go high until the delayed disable.

By switching from output to input and keeping the data register set to 0 there is no way the output stage should glitch.

ChrisMicro, 2018-02-07:
> @ChrisMicro @durin42 can either/both of you please try two independent things: 1) use different pins, eg 12 and 13 and re-run your test with the MPU6050; 2) apply the 1-line patch above and re-run with pins 5 and 4.

@dpgeorge 
I'm some weeks abroad and I have no hardware. Probably someone else could check it meanwhile.

peterhinch, 2018-03-08:
Have you tried using external pullups?

durin42, 2018-03-08:
To my (novice-level) understanding, my module already has external pullups. That said, I also tried adding those and it didn't help.

ChrisMicro, 2018-03-08:
@durin42: could you place somewhere the modified image? Because I have not installed a tool chain to recompile it myself.

durin42, 2018-03-08:
[micropython-patched-i2c.bin.zip](https://github.com/micropython/micropython/files/1794325/micropython-patched-i2c.bin.zip)

It's up to you if you trust me, I guess?

dpgeorge, 2018-03-08:
> Testing with pins 12 and 13 instead of 5 and 4 with released firmware also appears to work, but not as reliably: it's worked a couple of times, but also not worked a few times. Pins 12 and 13 are also unreliable for I2C with the patched firmware, in case that's interesting.

Thanks for testing and reporting back.  That is indeed interesting, that pins 12/13 act differently to 5/4.  It could be the particular board with its track routing, or lack of pull ups on these lines, or indeed something different in the esp8266 with how these lines are handled.

ChrisMicro, 2018-03-09:
@durin42 
Thank you very much the driver is working know very well on both interfaces
Here is my test code

```
from machine import I2C, Pin
import mpu6050
i2c = I2C(scl=Pin(5), sda=Pin(4))
accelerometer = mpu6050.accel(i2c)
for x in range(1,10000):                                                                  
  accelerometer.get_values()
```

with the driver from:
https://github.com/adamjezek98/MPU6050-ESP8266-MicroPython

ChrisMicro, 2018-03-09:
@dpgeorge 
Thank you very much for the patch. The SCL signal looks know as expected like with the Arduino driver and it works without my 12pF SCL-line hardware modification on both interfaces.
With my test above I could not detect a problem on pin 12/13 so I estimate that the problems on the I2C driver side are solved. I would think that possible issues on this pins are more likely to come from some hardware problems.

dpgeorge, 2018-03-12:
@ChrisMicro thanks for testing with the patch.  I have fixed the code in 033c32e694199bbb68816883a857a77711d68a13

closed 2018-03-12 by dpgeorge

ChrisMicro, 2018-03-12:
@dpgeorge Thank very much for fixing the problem and providing the patch.