micropython/micropython #3558: Setting HSE_VALUE in mpconfigboard.h leads to problems

rolandvs, 2018-01-10:
Depending on the board in `ports/stm32/boards/*` one could/would use a specific external crystal other than 8MHz for the hardware.

Setting `#define HSE_VALUE ((uint32_t)25000000)` in `mpconfigboard.h` is fine and everything compiles without a problem. However when using `pyb.freq()` or alternatively `machine.freq()` some strange values appear and things like baudrates of the UART's are set to a wrong value.

It seems that the `Makefile` (around line 455)  generates a header file generated by `pllvalues.py`. However this assumes that the `HSE_VALUE` is set like in the `stm32f7xx_hal_conf.h` and not elsewhere?!

dhylands, 2018-01-10:
The correct place to set HSE_VALUE is in stm32f4hxx_hal_conf.h. Setting it anywhere else may lead to issues.

dpgeorge, 2018-01-31:
@dhylands is correct: please define HSE_VALUE in stm32f7xx_hal_conf.h.

closed 2018-01-31 by dpgeorge