micropython/micropython #3559: ESP8266: how to remove compiled script?

hase-berlin, 2018-01-10:
I admit: this may be a user error. Still, I am confused.
Steps to reproduce:
- erase Flash on board
- flash MicroPython firmware
- start WebREPL
- upload a script (http_server.py)
- import the uploaded module
-- run the http server, use it
- delete the script (os.remove())
- reset the board; soft-reset/CRTL-D, Reset button or power-cycle
- reconnect to REPL (serial or WebREPL)
- import http_server

Imho, this last import should fail.
But it does not fail. Instead the deleted module is imported. And runs.

Uploading a modified version of the script then runs the old version - this is how I stumbled upon this behaviour the first place.
Effectively, I can not modify the http_server module that I originally uploaded interactively.

Erasing the Flash and flashing the firmware anew, works around the problem.
But if this is a bug and not a user error, it pretty much defies the purpose of an interactive access to the interpreter, imho.

slzatz, 2018-01-10:
Is it possible you have http_server.py in frozen modules?

hase-berlin, 2018-01-10:
nope.
Frozen modules are empty: Imade sure when reproducing he effect.

robert-hh, 2018-01-10:
After deleting the script, is it gone, when you execute the command:

import uos
uos.listdir()

hase-berlin, 2018-01-10:
import os
os.lisdir()
reports
['boot.py']
Just as before addind the script

robert-hh, 2018-01-10:
two questions:
- Which version of the firmware are you using (os.uname() result)?
- does that also happen if you do NOT use webrepl?
I cannot reproduce the behavior here, not using webrepl.

hase-berlin, 2018-01-10:
>>> os.uname()
(sysname='esp8266', nodename='esp8266', release='2.0.0(5a875ba)', version='v1.9.3-238-g42c4dd0 on 2018-01-09', machine='ESP module with ESP8266')

The reason I am using WebREPL is the file transfer capability. Any suggestions how to get by without that? A builtin kermit or xmodem would help...
Lets try manual file creation...
Nope. The auto-indent of the repl beats my nighttime skills. Will try again in the morning.

But thinking about the problem, I may have left out a step in the reproduction: I overwrote an existing file 'http_server.py' with a new version without deting it first. Maybe this triggered the observed behaviour.

Will report again tomorrow.

robert-hh, 2018-01-10:
You can use the adafruit ampy tool https://github.com/adafruit/ampy or dave hylands rshell https://github.com/dhylands/rshell for file transfer through USB. I also wrote an ftp server https://github.com/robert-hh/ESP8266-FTP-Server, but that is quite large and has to be embedded in flash, so it#s nothing to start with.

peterhinch, 2018-01-11:
I have never encountered this nor seen it reported in the forum. I routinely use rshell on USB to  overwrite files, issue ctrl-D and re-run.

robert-hh, 2018-01-11:
For rshell you migth have to set a small buffersize, like 
rshell --buffersize 20 ....

hase-berlin, 2018-01-11:
That did it! (with buffer-size instead of buffersize).
merci!