micropython/micropython #3560: ESP32: Init UART(3,9600)  creates errors

tcinbis, 2018-01-11:
Hey everyone, 
I am currently working with my ESP32 and tried to run the following code: 
```
>>> import machine
>>> from machine import UART
>>> uart = UART(3,9600)
E (22836) uart: uart_driver_delete(1084): uart_num error
E (22836) uart: uart_param_config(475): uart_num error
E (22836) uart: uart_driver_install(1007): uart_num error
E (22846) uart: uart_wait_tx_done(751): uart_num error
E (22846) uart: uart_set_baudrate(175): uart_num error
E (22856) uart: uart_get_baudrate(199): uart_num error
E (22856) uart: uart_set_pin(393): uart_num error
E (22866) uart: uart_set_pin(393): uart_num error
```

But unfortunately get this errors and I am not quite sure why?
I also ran the uart test for the pyboard to find out which uarts are available:
```
# test we can correctly create by id or name
for bus in (-1, 0, 1, 2, 3, 4, 5, 6, 7, "XA", "XB", "YA", "YB", "Z"):
    try:
        UART(bus, 9600)
        print("UART", bus)
    except ValueError:
        print("ValueError", bus)
```
And got the following output:
```
UART(1, baudrate=9600, bits=8, parity=None, stop=1, tx=10, rx=9, rts=-1, cts=-1, timeout=0, timeout_char=2)
UART: 1
UART(2, baudrate=9600, bits=8, parity=None, stop=1, tx=17, rx=16, rts=-1, cts=-1, timeout=0, timeout_char=2)
UART: 2
E (348156) uart: uart_driver_delete(1084): uart_num error
E (348166) uart: uart_param_config(475): uart_num error
E (348166) uart: uart_driver_install(1007): uart_num error
E (348176) uart: uart_wait_tx_done(751): uart_num error
E (348186) uart: uart_set_baudrate(175): uart_num error
E (348186) uart: uart_get_baudrate(199): uart_num error
E (348196) uart: uart_set_pin(393): uart_num error
E (348196) uart: uart_set_pin(393): uart_num error
E (348206) uart: uart_get_baudrate(199): uart_num error
UART(3, baudrate=393, bits=8, parity=None, stop=1, tx=0, rx=0, rts=-1, cts=-1, timeout=0, timeout_char=1)
UART: 3
```
So for my understanding uart 3 exists. But why can´t we use it? Is it reserved like uart 0?

Thanks in advance!

MaxBurgert, 2018-01-11:
I am getting the same error when I run your micropython code on my ESP32.

bmeisels, 2018-01-15:
To the best of my knowledge there is no uart 3 on the esp32.
Look at https://github.com/micropython/micropython/blob/master/ports/esp32/machine_uart.c

The code only mentions uart 0-2

tcinbis, 2018-01-15:
Yeah in the beginning I thought so too, but then I saw this output, which is kind of strange. 

`UART(3, baudrate=393, bits=8, parity=None, stop=1, tx=0, rx=0, rts=-1, cts=-1, timeout=0, timeout_char=1) `

After reading the esp idf documentation they also stated that only Uart 0-2 is available.

Is this maybe a software bug? Can someone else confirm this behavior?

dpgeorge, 2018-01-31:
Thanks for raising this issue.  UART(3) does not exist on the ESP32.  The error is simply that the check on the UART number was buggy and was accepting a value of 3.  That should now be fixed by 23f9f9495f0b64e70470b17ba38cf1f2148724bd

closed 2018-01-31 by dpgeorge