micropython/micropython #3561: [zephyr port]Issue when copy/pasting lenghty script in console paste mode

erwango, 2018-01-11:
Copy/pasting, when shell is in paste mode, fails with long scripts

For instance:
===         print("\nSending..14\n" + post_msg)
===         print("\nSending..15\n" + post_msg)
===         print("\nSending..16\n" + post_msg)
===         print("\nSending..17\n" + post_msg)
===         print("\nSending..18\n" + post_msg)
===        ~
            ~
              print("\nSending..~
                                 ~
                                  ~
                                   ~
                                    19\n" + post_msg)

As a consequence script could not be executed correctly.

Reproduced on Zephyr port, on both stm32 based and frdm_k64f boards.

The issue is systematic (will always happen at the same pasted line), but depends on the board (issue happens on different lines depending on the board)

erwango, 2018-01-11:
@pfalcon : can you have a look?
Could it be linked with zephyr driver implementation?

pfalcon, 2018-01-11:
@erwango: That `~` output means input buffer overflow. Unfortunately, input buffering in Zephyr doesn't work as good as in some other RTOSes I've seen. But anyway, a way to deal with that would be to increase the size of the buffer. I'll have a look what's currently used as the size there. You could help me by posting the full code chunk you're trying to paste.

erwango, 2018-01-11:
@pfalcon : ok, thanks for your response.

Above code was bogus (I was trying to understand behavior).

Code I was trying to copy/paste follows.
(I'd like to set up a webserver demo, eventually it should be longer)

import socket
import machine

#HTML to send to browsers
html = """<!DOCTYPE html>
<html>
<head> <title>ESP8266 LED ON/OFF</title> </head>
<center><h2>A simple webserver for turning HUZZAH Feather LED's on and off with Micropython</h2></center>
<center><h3>(for noobs to both the ESP8266 and Micropython)</h3></center>
<form>
LED0:
<button name="LED" value="ON0" type="submit">LED ON</button>
<button name="LED" value="OFF0" type="submit">LED OFF</button><br><br>
LED2:
<button name="LED" value="ON2" type="submit">LED ON</button>
<button name="LED" value="OFF2" type="submit">LED OFF</button>
</form>
</html>
"""

#Setup PINS
LED0 = machine.Pin(("GPIOB", 0), machine.Pin.OUT)
LED2 = machine.Pin(("GPIOB", 7), machine.Pin.OUT)

#Setup Socket WebServer
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind(('', 80))
s.listen(5)
while True:
    conn, addr = s.accept()
    print("Got a connection from %s" % str(addr))
    request = conn.recv(1024)
    print("Content = %s" % str(request))
    request = str(request)
    LEDON0 = request.find('/?LED=ON0')
    LEDOFF0 = request.find('/?LED=OFF0')
    LEDON2 = request.find('/?LED=ON2')
    LEDOFF2 = request.find('/?LED=OFF2')
    if LEDON0 == 6:
        print('TURN LED0 ON')
        LED0.off()
    if LEDOFF0 == 6:
        print('TURN LED0 OFF')
        LED0.on()
    if LEDON2 == 6:
        print('TURN LED2 ON')
        LED2.off()
    if LEDOFF2 == 6:
        print('TURN LED2 OFF')
        LED2.on()
    response = html
    conn.send(response)
    conn.close()

erwango, 2018-01-11:
By the way, I guess there should be a way with micropython to store/flash scripts so we can use multiple files and so on to do use more evolved scripts but I'm still looking for it. 
Is there a doc I can follow to do that?

pfalcon, 2018-01-11:
Well, that's the large paste content, it won't be easy to paste it in (maybe by chunks). I'll experiment later.

> By the way, I guess there should be a way with micropython to store/flash scripts

Yes, MicroPython supports concept of "frozen modules", similar to big Python. I recommend to use a branch from my repo described here: https://github.com/pfalcon/micropython/wiki/ZephyrSocket . Then you just drop Python modules here: https://github.com/pfalcon/micropython/tree/zephyr-wip/ports/zephyr/scripts , rebuild, reflash. As can be seen, that branch already contains couple of scripts which will be frozen, they're described in that wiki page.

tcinbis, 2018-01-11:
@erwango @pfalcon 
I don´t know anything about zephyr, but couldn´t you use ampy by adafruit [here](https://learn.adafruit.com/micropython-basics-load-files-and-run-code/install-ampy) to upload your code to the board and then start it via your shell?

pfalcon, 2018-01-12:
> I don´t know anything about zephyr

Filesystem is not available in Zephyr port.

pfalcon, 2018-01-12:
@erwango : So, currently input buffer size is set to 128 bytes: https://github.com/pfalcon/micropython/blob/pfalcon/ports/zephyr/prj_base.conf#L10 . That would be mean that as much as that could be pasted reliably. Of course, the expectation is that more than that can be pasted, because, well, that's a buffer, and an application processes it concurrently while it's filled.

With that note, let me test how that actually works on different boards.

erwango, 2018-01-12:
@pfalcon : Thanks, a least this is something I can play with.

pfalcon, 2018-01-12:
Ok, so testing paste on disco_l475_iot1, with 128 buffer size above, I can reliably paste 256 chars. In all fairness, that's good result. Going beyond that, to ~378 chars (128*3), I start to get mis-pastes: no "~" seen, but some trailer chars are lost. Again, I have a hunch that I saw better behavior in other systems, but again too, it's not too bad. Yes, would be nice to look if it can be improved, but given all the digging I did when it wasn't possible to reliably paste ~50 chars, and visible improvements since that, I'd leave it to someone else to lead this matter further ;-).

Summing up, to paste something really big, you'd need to paste it in chunks, bump buffer size, or better just freeze it into the binary as describe above.

erwango, 2018-01-12:
If pasting by chunks works, then this is a pretty viable solution and I recognize this was already something to achieve current status.
Anyway, freeze in binary will be more appropriate to my use, I'll check that.
Thanks for your help!

dpgeorge, 2020-06-09:
The zephyr port now also supports a filesystem so that's another way to get code running on it.  So I'll close this issue for now (pasting might even be improved with more recent Zephyr versions).

closed 2020-06-09 by dpgeorge