micropython/micropython #3562: STM32F407DiscBoard : Interfacing DHT22 sensor

nikhil1983, 2018-01-12:
Hello, 
So i am currently stuck on interfacing DHT22 sensor (without I2C) to stm disc board using micropython. I have source file written in C++ for interfacing DHT22 with stm board, and it works fine. It gives correct O/P on terminal. 

Using that C module as reference, I have made my python code for displaying sensor output on LCD using async method. But whenever i run the code, "DHT22 Not Responding " is getting printted on LCD. I don't know why my MCU is unable to read the data from sensor. I guess their is timing issues, but i cant see it. 

**Note**: Sensor is given supply voltage of 5 V and its working properly. 

I have tried to referred DHT22 drivers for esp28...,but it didn't helped either. If anyone have knowledge about this, please kindly help me out.  

Please have a look at my stm32f4_dht22.c and Python code ( .c file is just for reference) 

Source file written in C
> 
> > Source file : 
> > /******************************************************************//**
> > * @file	    stm32f4_dht22.c
> > * 
> 
> @brief    This is Temperature and Humidity sensor driver file
> * @version  V1.0
> * @date		Jan 24, 2017
> * @author	Bhavin Darji
> **********************************************************************/
> 
> 
> /* Includes ------------------------------------------------------------------- */
> #include "stm32f4_dht22.h"
> 
> /**
>  * @brief Initialize Pin For DHT22
>  * @param mode Pin mode output or input
>  *        @arg GPIO_MODE_INPUT
>  *        @arg GPIO_MODE_OUTPUT_PP
>  */
> void DHT22_PinConfig(uint8_t mode)
> {
>   GPIO_InitTypeDef  GPIO_InitStruct;
> 
>   /* Enable the GPIOC Clock */
> 
>   __GPIOC_CLK_ENABLE();
> 
>   /* Configure the GPIO_ALED pin */
>   GPIO_InitStruct.Pin = GPIO_PIN_6;
>   GPIO_InitStruct.Mode = mode;
>   GPIO_InitStruct.Pull = GPIO_PULLUP;
>   GPIO_InitStruct.Speed = GPIO_SPEED_HIGH;
> 
>   HAL_GPIO_Init(GPIOC, &GPIO_InitStruct);
> }
> /**
>  * @brief Initialize Pin for DHT22
>  * @return none
>  */
> void DHT22_Init(void)
> {
>   DHT22_PinConfig(GPIO_MODE_INPUT);
> }
> 
> 
> /**
>  * @brief Read DHT22 data and display it on UART4
>  */
> void DHT22_read(void)
> {
>     uint16_t   lastHumidity1;
>     uint16_t   lastHumidity2;
>     uint16_t   lastTemperature1;
>     uint16_t   lastTemperature2;
>     uint8_t    i, j, retryCount;
[... 305 lines cut ...]
> 
> #Calculate Check Sum
> 	csPart1 = currentHumidity >> 8
> 	csPart2 = currentHumidity and int(0xFF)
> 	csPart3 = currentTemperature >> 8
> 	csPart4 = currentTemperature and int(0xFF)
> 
> 
> 
> 	if (checkSum == ((csPart1 + csPart2 + csPart3 + csPart4) and int(0xFF))):
> 		lcd[0] = 'H1{}'.format(lastHumidity1)
> 		lcd[1] = 'T1{}'.format(lastTemperature1)
> 		await asyncio.sleep(1)
> 	else: 
> 		lcd[0] = 'Checksum Error'
> 		lcd[1] = ''
> 		await asyncio.sleep(1)
> 
> 
> 
> while True:
> 	loop = asyncio.get_event_loop()
> 	loop.run_until_complete(DHT22_read())

Please Help.

peterhinch, 2018-01-13:
This is the bug-tracker for MicroPython. You'll find a wider audience and so stand a better chance of getting help in [the forum](https://forum.micropython.org).

dpgeorge, 2018-01-31:
In commit efdda2c62deba4f4f24672e441cffe496158bb35 I added support for DHT sensors to the stm32 port, based on the code used by esp8266/esp32.  All stm32-based boards should now be able to use it as follows:
```python
from dht import DHT22
dht = DHT22(machine.Pin('X1'))
dht.measure()
print(dht.temperature(), dht.humidity())
```

closed 2018-01-31 by dpgeorge