micropython/micropython #3565: Performance Question

notzoom, 2018-01-14:
If embedded systems need fine tuning for good performance, how can Micropython not use C data types ? Thanks.

ryannathans, 2018-01-15:
I don't know what you're looking for but maybe the uctypes module is what you want?

dpgeorge, 2018-01-31:
The recommended way to get improved performance for a particular subsystem is to implement it in C, and then expose it as a module in MicroPython.  uctypes may help to pass data between C and MicroPython code.  For examples of modules written in C see `drivers/dht/dht.c` (and corresponding `drivers/dht/dht.py`) and also `extmod/modonewire.c`.

Otherwise, these kinds of questions are best asked at https://forum.micropython.org

closed 2018-01-31 by dpgeorge