micropython/micropython #3566: Variables in functions do not exist in locals() - breaking exec

ryannathans, 2018-01-15:
Possibly related to #2833
Found on Unix port
Declaring a variable in a function and printing locals() inside that function will not show the created variable, unlike CPython.

```
def test():
    a = 42
    print(locals())
```

peterhinch, 2018-01-15:
See [differences](https://github.com/micropython/micropython/wiki/Differences).

ryannathans, 2018-01-15:
Ah, missed that point 8 when skimming the page prior to reporting.

What is the purpose of the locals function in micropython if it will almost never have the locals in it?

dpgeorge, 2018-01-31:
> What is the purpose of the locals function in micropython if it will almost never have the locals in it?

It's valid (and sometimes useful) inside the body of a class definition, eg:
```python
class C:
    print(locals())
```

closed 2018-01-31 by ryannathans