micropython/micropython #3567: Minimal ports compiles but does not run

Tobias-Kohn, 2018-01-15:
I have tried to compile and run the minimal port on three different Linux systems. Compiling the source code using `make` works without any problems. I can then execute the file `build/firmware.elf` and get the usual Python prompt. However, when I enter anything, it is just echoed back and the program does not respond anymore. It looks like this (the `^C` coming from my terminating the application):
```
MicroPython v1.9.3-240-ga275cb0 on 2018-01-15; minimal with unknown-cpu
>>> 3+4
3+4^C
```
I get this on both CentOS and Ubuntu systems (as the system does not detect the cpu: it's Intel i5/i7, x86-64). Am I missing something there, or is there something broken?

robert-hh, 2018-01-15:
Maybe it's just the line end detection. Did you try to end the line with Ctrl-J instead of Enter, which is Ctrl-M?

Tobias-Kohn, 2018-01-15:
I have just tried `Ctrl-J`, but, unfortunately, I can't see any difference in the behaviour.
Thanks for the suggestion, though.

pfalcon, 2018-01-15:
When something didn't work 3 times, time to read the documentation.

Tobias-Kohn, 2018-01-15:
@pfalcon Which part of the documentation are you referring to? It seems like most of the documentation is meant for the PyBoard. I am also aware that there is a Unix-Port (which works well on my machines).
But referring to both the general README and the one included in `ports/minimal`, this should be the place to start when one wants to build a new port/system.
But if you have some specific part of the documentation in mind, a pointer to it would be appreciated.

pfalcon, 2018-01-15:
Right this line: https://github.com/micropython/micropython/blame/master/ports/minimal/README.md#L14 .

Tobias-Kohn, 2018-01-15:
That worked, thanks!
I misunderstood the documentation in that I thought `make run` would run a test-suite or something like that. Perhaps, you might want to slightly change the wording, as I really did not run a test script but rather the REPL.
Anyway, thank you for your help!

closed 2018-01-15 by Tobias-Kohn

pfalcon, 2018-01-16:
Right, there're glitches in docs, however "make run" is there, and should give enough hints ;-). Can also be verified via *executable* documentation (e.g. Makefile), which never lies: https://github.com/micropython/micropython/blob/master/ports/minimal/Makefile#L79 .

I've fixed docs in my fork: https://github.com/pfalcon/micropython/commit/064e84708d567ec60eadb45fa38c4af38bcec92f

dpgeorge, 2018-01-31:
Docs have been updated in 524ff3027525832c85bf5411a6302578836b941a to better describe what "make run" does.