micropython/micropython #3568: setattr creating duplicates for existing attributes

Morteo, 2018-01-16:
Why does **setattr** behave differently on micropython to python3 ?

File: intropsect.py
```
#!/usr/bin/python3

class A():
  one = "1"
  two = "2"
  def x():
    pass

class B():
  two = "ii"
  three = "iii"
  def y():
    pass

for attr in dir(a):
  if not attr.startswith('__') and not callable(getattr(a,attr)):
    print("a=",attr,getattr(a,attr))
    setattr(b,attr,getattr(a,attr))

for attr in dir(b):
  if not attr.startswith('__') and not callable(getattr(b,attr)):
    print("b=",attr,getattr(b,attr))
```

pi@raspberrypi:~/working $ ampy --port /dev/ttyUSB0 run ./temp/introspect.py
```
a= one 1
a= two 2
b= three iii
b= two 2
b= two 2
b= one 1
```
pi@raspberrypi:~/working $ ./temp/introspect.py
```
a= one 1
a= two 2
b= one 1
b= three iii
b= two 2
```

stinos, 2018-01-16:
Your code has errors, I assume `a` has to be `A`? If I change it like that both CPYthon and MicroPython (built from latest source) give the same result. So, you should state what version you are using. And try to update as this might have been fixed already.

peterhinch, 2018-01-16:
I think what we're seeing here is a difference between the output of `dir()` rather than a problem with `setattr`. If you paste the following
```python
class B():
    two = "ii"
    three = "iii"
b = B()
setattr(b, 'two', 2)
```
Then issue
```
>>> b.two
2
>>> B.two
'ii'
```
This conforms with CPython. `setattr(B, 'three', 3)` also works as expected, setting the class variable. Where behaviour differs is MicroPython's
```
>>> [x for x in dir(b) if not x.startswith('__')]
['two', 'three', 'two']
>>> 
```
IOW MicroPython's `dir()` lists the class variable as well as the instance variable with the same name. CPython's lists class and instance variables but in the case where both have the same name, the name appears once only. It's not obvious to me which is correct or if it much matters.

dpgeorge, 2018-03-07:
It does look like this was an issue with `dir()` rather than `setattr()`, and since 98647e83c733a8a8d36522f71e677367ca8ddd03 `dir()` was improved so that it doesn't list duplicates, so this issue is now fixed.

closed 2018-03-07 by dpgeorge