micropython/micropython #3570: [ESP8266] Firmware OTA update

zhakhalov, 2018-01-16:
Hello,

I'm working over OTA update of the device using firmware files.
The process is following
- Download file to filesystem
- Somehow verify it
- Write downloaded file to the start of the flash

I'm done with first two steps but I'm facing issue with writing firmware to flash. I see that flash and firmware file are almost the same except very first parts.

I hope you can point me at wich point I should write file to the flash.
I mean what are offset in bytes from the 0 address.

Thanks in advance.

stinos, 2018-01-16:
This is the issue tracker for reporting bugs, feature requests etc. For general questions about MicroPython please use https://forum.micropython.org

zhakhalov, 2018-01-16:
Thanks for referring me to forum but I'm under impression that I would not find this information there.

I would like to contribute this feature to micropython but I need some assistance with understanding of flash layout.

robert-hh, 2018-01-16:
Did you notice that OTA is already implemented for the ESP8266? On teh download page http://micropython.org/download you'll find the respective information.

zhakhalov, 2018-01-16:
The way OTA is implemented now is not the best. 

Thank you for your time.
Ticket can be closed

robert-hh, 2018-01-16:
You can (should) close it yourself.

closed 2018-01-16 by zhakhalov