micropython/micropython #3573: Stack-smashing protection causes mpy-cross to segfault

Cheaterman, 2018-01-17:
Hi!

As mentioned in the title, hardened GCC will prevent `mpy-cross` from working (segfaults in `nlr_push()`). A workaround is to use `export CFLAGS_MOD='-fno-stack-protector'` before compiling `mpy-cross`. I don't know if that's an acceptable fix for upstream though so I didn't make a PR.

Thanks!

Cheaterman, 2018-01-17:
(More details are available in https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/19 and https://github.com/micropython/micropython/issues/3526 )

dpgeorge, 2018-01-31:
Please try using the setjmp implementation of NLR handling, instead of the native one, and see if that fixes it (keep stack protector enabled).  You can do this by adding the line `#define MICROPY_NLR_SETJMP (1)` to mpy-cross/mpconfigport.h and then rebuilding.

jonnor, 2024-09-19:
Is this still an issue in recent MicroPython (version 1.23 or later)? Does the above suggestion fix the issue?

jonnor, 2025-03-03:
No feedback in 6 years

closed 2025-03-03 by jonnor