micropython/micropython #3574: ure: RuntimeError : maximum recursion depth exceeded When length string is over 129

DomCharn, 2018-01-17:
in case of string length exceeding 129 caracters RunTime Error appears

import ure
A=b'{_toghh:I:{12345678901234567890123456789013456789012345678901234567890123456789012345678901234567890123456789012345678901234567}}'
 len(A)
129
 R=ure.search(b':I:{(.*)}',A)
 R.group(1)
b'12345678901234567890123456789013456789012345678901234567890123456789012345678901234567890123456789012345678901234567}'
A=b'{_toghh:I:{123456789012345678901234567890134567890123456789012345678901234567890123456789012345678901234567890123456789012345678}}'
len(A)
130
R=ure.search(b':I:{(.*)}',A)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: maximum recursion depth exceeded

DomCharn, 2018-01-19:
In my haste was omitted : 19.2  official version Esp8266

robert-hh, 2018-01-19:
The example above seems to be wrong. I suppose it should be:
`R=ure.search(b':I:{(.*)}',A)`
Besides that: there were already a few cases, where this error was observed with ure. I would not see that as a bug, more as a consequence of limited resources.

dpgeorge, 2018-01-31:
Yes, @robert-hh is correct: this is just a limitation of the way ure is implemented combined with limited resources (ie little RAM).  To workaround it try rewriting the regex, or splitting the input string into pieces before searching it with the regex.

Otherwise, this is essentially a duplicate of #2451.

closed 2018-01-31 by dpgeorge