micropython/micropython #3579: no module named 'webrepl_setup' ESP32

katjap, 2018-01-23:
Uploaded esp32-20180122-v1.9.3-240-ga275cb0f.bin
I'm able to connect to serial REPL, but when I attempt to import webrepl_setup it raised an ImportError

```
MicroPython v1.9.3-240-ga275cb0f on 2018-01-22; ESP32 module with ESP32
Type "help()" for more information.
>>> I (3971) wifi: pm start, type:0

import webrepl_setup
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'webrepl_setup'

```
is there no webrepl for ESP32?

robert-hh, 2018-01-23:
There no webrepl for ESP32!

katjap, 2018-01-23:
Bummer. I'll have to figure out OTA then. I kinda figured this was the case after I ran help('modules'). I hope it gets ported soon.

closed 2018-01-23 by katjap

dpgeorge, 2018-05-02:
WebREPL on ESP32 is now implemented.

Sniphs98, 2023-06-30:
@dpgeorge 
Hello there! I was wondering if Lib is currently available for Raspberry Pi Pico or if there are any plans to introduce it in the future? Thank you for your help!