micropython/micropython #3585: ESP-WROOM-32 - no respond

ds2k5, 2018-01-28:
Did not respond to keyboard

ESP-32 (ESP-WROOM-32)

**esptool.py --port /dev/ttyUSB0 erase_flash**
esptool.py v2.2.1
Connecting........_
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.6s
Hard resetting...

**esptool.py --chip esp32 --port /dev/ttyUSB0 write_flash -z 0x1000 esp32-20180128-v1.9.3-240-ga275cb0f.bin**

**minicom -b 115200 -D /dev/ttyUSB0**

Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyUSB0, 06:32:10

Press CTRL-A Z for help on special keys                                                    
                                                                                           
I (345) cpu_start: Pro cpu up.                                                             
I (345) cpu_start: Single core mode                                                        
I (346) heap_init: Initializing. RAM available for dynamic allocation:                     
I (349) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (355) heap_init: At 3FFDCD68 len 00003298 (12 KiB): DRAM
I (361) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (368) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (374) heap_init: At 4008FC7C len 00010384 (64 KiB): IRAM
I (380) cpu_start: Pro cpu start user code
I (174) cpu_start: Starting scheduler on PRO CPU.
OSError: [Errno 2] ENOENT
MicroPython v1.9.3-240-ga275cb0f on 2018-01-28; ESP32 module with ESP32
Type "help()" for more information.
>>> 

Looks like the boot.py is empty

**ampy --port /dev/ttyUSB0 get boot.py**
# This file is executed on every boot (including wake-boot from deepsleep)

robert-hh, 2018-01-28:
The fault is minicom. I just tried minicom too, it did not work. I am using picocom.

ds2k5, 2018-01-28:
thanks robert-hh,
did work for me too with picocom

robert-hh, 2018-01-28:
That's fine, and have fun with Micropython. may I ask you to close the issue then:?

closed 2018-01-29 by ds2k5

ChrisMicro, 2018-03-14:
Minicom is working if you turn of the hardware handshake in minicom:

CTRL-A => Z => O => "serial setup" => F