micropython/micropython #3586: Backport CircuitPython GC changes in to uPy

nevercast, 2018-01-28:
Hello,

This issue is for the discussion of bringing CircuitPython's GC changes, [commit](https://github.com/adafruit/circuitpython/commit/416abe33ed44e44d8f75342731080217e123278b), example [video](https://www.youtube.com/watch?v=gSfmVdRejTk) in to MicroPython.

@tannewt

This GC change shows a good improvement over fragmentation by separating long-lived and short-lived objects to opposite ends of the heap. Fragmentation is a long-standing issue with MicroPython and I'm hoping that bringing Scott's changes in to uPy will alleviate some of the memory stress. There has already been several discussions about changes to garbage collection and memory allocation, those I found immediately are: #2057 #1168 #1161. Mentions for members involved: @dpgeorge @pfalcon @danicampora

We should find an approach forward that does not block any ongoing progress towards GC/MM improvements. I'm hoping that the members mentioned can help clarify what is currently in progress, and what is open for contribution.

tannewt, 2018-01-29:
@aykevl Also has a related PR open #3580.

I'm happy to help merge. The PR should be well isolated already.

adritium, 2018-05-31:
Allocating on lifetime is hard: how is a C-module supposed to guess how whether the object is going to be long- or short-lived? Maybe with `.py` hints #3774 but that'll get messy and unmaintainable. 

Why not divide the allocated objects according to size? S[plit up heap into logical areas based on allocated size.](https://stackoverflow.com/a/2276452/4784683)