micropython/micropython #3588: Refactor use of MICROPY_QSTR_EXTRA_POOL to clarify its relationship with frozen_mpy

adritium, 2018-01-30:
`MICROPY_QSTR_EXTRA_POOL` is related to the use of `frozen_mpy` code i.e. the transformation `.py`->`.mpy`->`.c` which is used to precompile `.py` on the PC - enabling compilation of "infinitely" large `.py` 0 and to execute code from flash instead of RAM.

However, unless I'm greatly misunderstanding the purpose of `MICROPY_QSTR_EXTRA_POOL`, this relationship could be greatly clarified in the code.

For example, why not place `#ifndef MICROPY_QSTR_EXTRA_POOL ...` inside this `#ifdef`?

https://github.com/micropython/micropython/blob/a275cb0f487cd6517760271dc01d369c32600c63/py/mpconfig.h#L629-L631

Only reason would be if `MICROPY_QSTR_EXTRA_POOL` is not solely related to `frozen_mpy`.

https://github.com/micropython/micropython/blob/a275cb0f487cd6517760271dc01d369c32600c63/ports/minimal/mpconfigport.h#L1-L11

https://github.com/micropython/micropython/blob/a275cb0f487cd6517760271dc01d369c32600c63/py/qstr.c#L98-L126

Portion of a generated `frozen_mpy.c`

```
extern const qstr_pool_t mp_qstr_const_pool;
const qstr_pool_t mp_qstr_frozen_const_pool = {
    (qstr_pool_t*)&mp_qstr_const_pool, // previous pool
    MP_QSTRnumber_of, // previous pool size
    5, // allocated entries
    5, // used entries
    {
        (const byte*)"\xfa\x11" "frozen_py/main.py",
        (const byte*)"\x68\x03" "led",
        (const byte*)"\xa9\x07" "main.py",
        (const byte*)"\xb8\x08" "main1.py",
        (const byte*)"\x1b\x08" "main2.py",
    },
};
```

dpgeorge, 2018-02-07:
> Only reason would be if MICROPY_QSTR_EXTRA_POOL is not solely related to frozen_mpy.

Yes that's the reason: MICROPY_QSTR_EXTRA_POOL is a general way to allow a port to provide an extra pool of qstrs which is outside the automatically generated one.

dpgeorge, 2018-02-26:
I assume this point was clarified.

closed 2018-02-26 by dpgeorge