micropython/micropython #3589: uctypes aggregate arrays ignore definition offsets and instead are padded by largest type in definition

ryannathans, 2018-01-31:
This doesn't seem like intended behavior, the offsets in the array definition seem to be ignored.

This was found on the unix port.

In this snippet of code I would expect an aggregate array four long, each with 12 bytes (UINT32 + UINT64)
```
buffer = bytearray(48)
definition = (uctypes.ARRAY, 4, {'element_1': uctypes.UINT32 | 0, 'element_2': uctypes.UINT64 | 4})
struct = uctypes.struct(uctypes.addressof(buffer), definition)
print(buffer)
struct[3].element_1 = 555
print(buffer)
```

Output:
```
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
```

Where did my bytes go? Off the end it seems... Into uncharted territory...

Writing to the second element_1

```
struct[1].element_1 = 555
print(buffer)
```

Output:
`bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00+\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')`

That's in the wrong spot! I expected this to modify the 13th + 14th byte but it seems the UINT32s are padded as UINT64s and instead modified the 17th + 18th byte, totally ignoring my offsets (0 and 4) in the array definition.

This padding behavior does not occur on non-aggregate structs
```
buffer = bytearray(20)
definition = {'element_1': uctypes.UINT32 | 0, 'element_2': uctypes.UINT64 | 4, 'element_3': uctypes.UINT32 | 12}
struct = uctypes.struct(uctypes.addressof(buffer), definition)
print(buffer)
struct.element_3 = 555
print(buffer)
```

Output:
```
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00+\x02\x00\x00\x00\x00\x00\x00')
```
That's the right spot! The 13th and 14th byte.

tldr: aggregate arrays ignore offsets in definition and instead choose to pad all elements by the largest type which makes the module useless for user defined definitions of aggregate arrays.

dpgeorge, 2018-02-07:
> This doesn't seem like intended behavior, the offsets in the array definition seem to be ignored.

I think it is intended.  The offsets are honored for individual items in the array, but the point is that *each item is aligned to the alignment requirements of the biggest field*.  The code explicitly does this here: https://github.com/micropython/micropython/blob/master/extmod/moductypes.c#L257-L260

> which makes the module useless for user defined definitions of aggregate arrays.

This alignment behaviour mimics how C does it with arrays.  The idea of uctypes is that you can create a description of a struct in memory that aligns exactly with the corresponding C type.

If you have an array with unaligned data then it would be better to represent it in a different way, not with UINT64 types.

ryannathans, 2018-02-07:
When creating a struct in C, isn't the default behavior to align to the machine word size, not the biggest data type in the struct?

dhylands, 2018-02-07:
In C, each field is aligned to the alignment of the field. The structure as a whole is aligned to the largest alignment of each field. This way, each field is aligned properly in each element of an array of structs. So the following structure:
```
struct {
  uint8_t a;
  // 1 byte of padding here
  uint16_t b;
  uint32_t c;
  uint8_t d;
  // 3 bytes of padding here
};
```
The sizeof the struct will be 12, a will be at offset 0, b at 2, c at 4, and d at 8.

Going back to the original post, this struct:
```
struct {
  uint32_t a;
  // 4 bytes of padding here
  uint64_t b;
  uint32_t c;
  // 4 bytes of padding here
}
```
should have a size of 24, with a at offset 0, and b at offset 8, and c at offset 16.

The only way in C to get b at an offset of 4 is to use a packed structure.

ryannathans, 2018-02-07:
Today I learned that fields in C structs are padded to align the following element (where I thought they were padded to the machine word size), and the final element is padded to align the struct to the size of the largest element.

ryannathans, 2018-02-27:
How would one work with packed structs in an aggregate array if the definition is ignored? i.e __attribute__((packed))

ryannathans, 2018-02-27:
Turns out, you can work with packed data structures by specifying a non-native layout type. This is in the documentation, I had just missed it. LOL

closed 2018-02-27 by ryannathans