micropython/micropython #3592: Assigning to __class__ fails without error

nickovs, 2018-02-02:
In the standard Python implementation it is possible to assign to the `__class__` attribute of an object. This means that the following code works:
```
class A:
    def __init__(self, x):
        self.x = x

class B(A):
    @property
    def y(self):
        return self.x*2

a = A(10)
a.__class__ = B
print(a.y)   # Prints 20
```
One can debate the desirability of this sort of code but it can be quite useful for cases such as unpacking structured data where part of the structure contains type information and you'd like to have different accessor methods for the fields depending on the type tag.

Anyway, whether you like this style or not, in MicroPython assigning to `__class__` silently fails. While I'd personally prefer it to succeed at the very least if it's going to fail it should fail with a `AttributeError` saying that the attribute can't be set.

nickovs, 2018-02-05:
Thanks for the reply. I had started to look into an implementation and concluded that it wasn't too hard as long as the new class inherits from the same built-in class as the old one (and the target is a proper class, not a small value). Swizzling a class to a new class that inherits from a different built-in potentially leads to illegal memory accesses and should probably be prevented.

I can try to write a patch if you'd like but I suspect that reviewing my pull request might you take as long as writing the code :-) Let me know if you want me to have a go.

dpgeorge, 2018-02-06:
I provided a working patch in #3599.

> Swizzling a class to a new class that inherits from a different built-in potentially leads to illegal memory accesses and should probably be prevented.

I didn't think to check for this... you are right that leaving such a hole open (ie not checking for the same base builti-in) will lead to potential problems.  It's a bit of effort to do such a check.  I wonder how CPython handles this case...

nickovs, 2018-02-06:
It looks like the [relevant CPython code](https://github.com/python/cpython/blob/master/Objects/typeobject.c#L3909) restricts assignment of `__class__` to heap types and classes made using ModuleType. There is a comment in the code that says that they tried to do something much more clever in 3.5 but it had a subtle bug so they reverted it until they could find a better solution.

For the vast majority of use cases the restriction the CPython 3.6 has is rarely a problem so you could just do the same. From a memory point of view it isn't hard to allow more but you end up with failed assumptions about immutability and that way lies subtle bugs.

nickovs, 2018-02-06:
In CPython as well as restricting the assignment to classes that are built at runtime, the function [compatible_for_assignment](https://github.com/python/cpython/blob/master/Objects/typeobject.c#L3864) checks to see if the both the old and new classes are have the same memory size as the nearest common ancestor. The idea being that if a descendant has the same size as its ancestor then it is memory-compatible since it can't have added any new fields.

As far as I can tell in MicroPython there is no way to find out at runtime the memory size of a built-in class, since built-in types get allocated using macros that ultimately expand out to `m_malloc(sizeof(type))`. Fixing this would need an extra field in `struct _mp_obj_type_t` and add an overhead of a size_t per type, which would add a few hundred bytes just for the built-ins, with very little benefit.

I suspect that the best option is going to be to restrict assignment to non-native classes which descend from the same native type. Walk up each of the two lists of parents until you hit something for which `mp_obj_is_native_type()` is true and then see if they are the same. This makes the assumption that all native types are memory-incompatible with each other; I assert that that is a reasonable assumption since any exceptions to it are implementation details that should be hidden from the user. It's also fairly easy to do and results in behaviour that would appear to be the same as CPython.

dpgeorge, 2018-02-07:
> I suspect that the best option is going to be to restrict assignment to non-native classes which descend from the same native type. Walk up each of the two lists of parents until you hit something [...]

Yes, this is a fair restriction and assumption and would work well in practice.

But, it now gets to the point that to implement this feature cleanly and correctly would require a non-trivial amount of additional code.  And for a feature that is arguably not often used.

If it's possible to find a way to implement the feature in a minimal way then it can be considered for inclusion.  Otherwise, if it requires more code, it would be better to wait and see if more users find this feature desirable and worth the the cost.

nickovs, 2018-02-07:
I think you are right. In the short term I think that the best way to close this issue would be for attempts to assign to `__class__` to raise an exception. Separately, I will think some more about an efficient and reliable way to do do this properly and if I can come up with a patch that only adds a few dozen bytes to the STM32 image size I'll submit a pull request.

nickovs, 2018-02-07:
So I took a closer look and it seems like all of the necessary code to do this properly already exists in `objtype.c`. The existing `instance_count_native_bases()` function returns the underlying native class of a type, so you can just call this on the current value of `self->base.type` and the proposed new value and see if they return the same thing.

I have whipped up a [patch](https://github.com/nickovs/micropython/commit/70b2dc5c4dbc440db2b98f1f2d07937fcc38b3f2) based on your code with this one extra check. It adds 88 bytes to the STM32 image (36 bytes more than your patch). If you think that this is worth pursuing then I will clean it up, add some comments and write some test code. If you think that 88 bytes (iff `MICROPY_CPYTHON_COMPAT` is defined) is too much then we can just have attempts to set `__class__` raise an exception (although to be fair applying this check and raising the exception will probably take 40+ bytes!)

dpgeorge, 2018-09-20:
@nickovs thanks for making the patch.  I think the code you wrote is still valid even after the recent optimisation to improve the case when there are no special accessors in a type.  But I'm unsure if it's wise to include this because it's one of those really dynamic features of Python which might prevent future optimisations (eg static compilation, caching of methods, and JIT-like optimisations).  Unless assigning to `__class__` has some really compelling use cases I'd rather not support it, so users find a way to work without it, leaving the way open for possible optimisations.