micropython/micropython #3593: DHT22 no result

ds2k5, 2018-02-03:
Hello,
try to get Temprature from a DHT22

Board: NodeMcu V3, ESP8266
Board      DHT22
GND -->  right (4)
V3 -->      left (1)
PIN4 -->  Signal (2)
```

OPTIONS: I18n 
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyUSB0, 23:15:14

Press CTRL-A Z for help on special keys

>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== 
=== import dht
=== import machine
=== import time
=== d = dht.DHT22(machine.Pin(4))
=== d.measure()
=== d.temperature()
=== d.humidity()
=== 
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
  File "dht.py", line 13, in measure                                            
OSError: [Errno 110] ETIMEDOUT                                                  
>>> 

```

chrisb2, 2018-02-03:
According the [datasheet](https://www.sparkfun.com/datasheets/Sensors/Temperature/DHT22.pdf), see section 6; "When power is supplied to sensor, don't send any instruction to the sensor within one second to pass unstable status", perhaps this is the cause of the issue?

regards,
Chris

ds2k5, 2018-02-04:
Thanks Chris,
but got error with your CODE
```

paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import Pin
=== from utime import sleep_ms
=== import dht
=== 
=== sleep_ms(1000)
=== 
=== sensor = dht.DHT22(Pin(4))
=== while True:
===     sensor.measure()
===     print("%0.fC, %0.f%%" % (sensor.temperature(), sensor.humidity()))
===     sleep_ms(1000)
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
  File "dht.py", line 13, in measure
OSError: [Errno 110] ETIMEDOUT

```

ds2k5, 2018-02-04:
Is it possible to see if a sensor: dht-22 connected ?

from machine import Pin
import onewire
ow = onewire.OneWire(Pin(4))
ow.scan()   

Output: []

closed 2018-02-05 by dpgeorge