micropython/micropython #3594: enable UDP callback

tschudin, 2018-02-05:
In [this forum post](https://forum.micropython.org/viewtopic.php?t=3101), a patch was presented to enable callbacks for UDP sockets, which would be very helpful for simple UDP-based servers. Will this be included?

```
--- a/extmod/modlwip.c
+++ b/extmod/modlwip.c
@@ -281,6 +281,7 @@ STATIC void _lwip_udp_incoming(void *arg, struct udp_pcb *upcb, struct pbuf *p,
         socket->incoming.pbuf = p;
         socket->peer_port = (mp_uint_t)port;
         memcpy(&socket->peer, addr, sizeof(socket->peer));
+       exec_user_callback(socket);
     }
 }
```

cefn, 2018-02-17:
I too would be interested in this functionality. 

In particular I am curious if https://github.com/jczic/MicroDNSSrv can be ported to ESP8266 Micropython (where there is no _thread module) by using set_sock_opt() to wire up a handler.

This would mean that, for example, a captive portal could be created which causes all DNS to resolve to 192.168.4.1 or which allows a more memorable and less confusing URI (like http://iot ) to be shared with users of a self-contained wifi-controllable device (versus the complexity of LAN setup and extracting IP addresses to make things addressable as things stand).

jonnor, 2024-09-29:
I believe this is still missing in latest MicroPython