micropython/micropython #3595: SSL Socket wrap doesn't flush underlying socket as expected

cefn, 2018-02-05:
I have been experiencing an issue with Twitter SSL data over HTTPS, (presumably because the Twitter server doesn't close the socket triggering EOF). 

In micropython, the SSL socket doesn't seem to receive all the expected data, while in python3 all the data arrives. I have been testing in unix to check that I am in an environment with plenty of memory available for TLS records.

You can see the thread on the forum summarising the findings so far, including (in the last post in the thread) a self-contained script which should replicate the problem without needing to log onto get a Twitter developer account, although the bearer token in the script should be revoked shortly to minimise potential abuse ...

https://forum.micropython.org/viewtopic.php?f=2&t=4356#p25230

dpgeorge, 2018-02-07:
Thanks for the report.  I replied on the forum in that topic.  The issue is that MicroPython's ssl.read(n) method will try to read exactly n bytes so waits forever if there is not enough data.  The solution is to parse the content-length header and only read that many bytes of data.

cefn, 2018-02-14:
Thanks hugely for this. 

Now have all working on CPython and ESP32 but outstanding bug on ESP8266 related to low memory and graceless reset when calling ssl socket wrap to be filed separately.

closed 2018-02-14 by cefn