micropython/micropython #3597: Implement ujson.dump for memory efficient stream dumps

orzechow, 2018-02-06:
For the use case of dumping jsons to a file or any other io stream (when would you dump a json otherwise?), the usage of `ujson.dumps()` imposes the necessity of enough memory to store the parsed string:

```python
config_file = open("config.json", "w")
config_file.write(ujson.dumps(config))
config_file.close()
```

I would like to see a way of dumping directly to an IO stream as with CPython's `json.dump()` function, that takes a `.write()` capable stream.
So, something like:

```python
config_file = open("config.json", "w")
ujson.dump(config, config_file)
config_file.close()
```

@dpgeorge mentioned in #2489 that this should not be a lot of work. So I would be glad to see it coming :)

dpgeorge, 2018-02-06:
Please see #3598 for an implementation.  If you can test it that would be great!

dpgeorge, 2018-02-15:
Ok, `ujson.dump()` was implemented in d9bca1f7bddad762aac72b0fcbdd84eb22e1c3c1

closed 2018-02-15 by dpgeorge

orzechow, 2018-03-06:
Works like a charm, thanks!