micropython/micropython #3600: Unwanted charter "Â" in HTML Output - DHT22 + ESP8266

ds2k5, 2018-02-06:
Hello,
hat a unwanted charter "Â" in the Output between Temrature and °C

found this sample Code:

How to fix this ?

```
import machine
import dht
import socket

def do_connect():
    import network

    SSID = 'MyWLAN01'
    PASSWORD = 'MyWLANPassw0rd'

    sta_if = network.WLAN(network.STA_IF)
    ap_if = network.WLAN(network.AP_IF)
    if ap_if.active():
        ap_if.active(False)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.active(True)
        sta_if.connect(SSID, PASSWORD)
        while not sta_if.isconnected():
            pass
    print('Network configuration:', sta_if.ifconfig())

do_connect()

CONTENT = b"""\
HTTP/1.0 200 OK

<h1>Micropython weather station - ESP8266 with DHT-22</h1>
<p>Temperature: {0}°C</p>
<p>Humidity: {1}%</p>
"""

# Wemos D1 Mini DHT Shield is on pin 2 (D4)
# Use ESP8266 GPIO pin numbers
# https://www.wemos.cc/product/d1-mini-pro.html
d = dht.DHT22(machine.Pin(4))

def main():
    s = socket.socket()
    addr = socket.getaddrinfo("0.0.0.0", 80)[0][-1]

    s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)

    s.bind(addr)
    s.listen(1)
    print("Listening, connect your browser to http://addr:80/")

    counter = 0
    while True:
        res = s.accept()
        client_s = res[0]
        client_addr = res[1]
        req = client_s.recv(1024)

        print(req)

        d.measure()

        t = d.temperature()
        h = d.humidity()

        client_s.send(CONTENT.format(t, h))
        client_s.close()
        counter += 1
        print()

main()
```

Output:

Micropython weather station - ESP8266 with DHT-22

Temperature: 26.0Â°C

Humidity: 28.2%

 
Page source:
 
```
<h1>Micropython weather station - ESP8266 with DHT-22</h1>
<p>Temperature: 26.0Â°C</p>
<p>Humidity: 29.0%</p>
 
```

pmp-p, 2018-02-06:
you did not set encoding, so better replace ° by &deg;

ds2k5, 2018-02-06:
Thanks a lot this works fine!

closed 2018-02-06 by ds2k5