micropython/micropython #3602: Simple http server issue on esp8266

Sunrise17, 2018-02-06:
I am sending same post to Simple Http Server(Micropython) installed on ESP8266 and BaseHTTPServer(Python) on PC.

On PC, i am getting Payload successfullya as seen below, However, when i post to ESP8266, i am not seeing any payload in the body and after 3-5 request, i am getting an following error; you can see the code in ESP8266 below;

```
2018-02-06 15:58:48.381801+0300 Chroma[1655:40174] TIC TCP Conn Failed [49:0x60400016c6c0]: 1:61 Err(61)
2018-02-06 15:58:48.385161+0300 Chroma[1655:40174] Task <542713C7-815D-479B-A462-DBD6A6EBB795>.<49> HTTP load failed (error code: -1004 [1:61])
2018-02-06 15:58:48.386808+0300 Chroma[1655:40159] Task <542713C7-815D-479B-A462-DBD6A6EBB795>.<49> finished with error - code: -1004
Could not connect to the server.
```

PC-DEBUG;
```
----- Request Start ----->
Request path: /
Content Length: 20
Request headers: Host: 0.0.0.0:10000
Content-Type: application/x-www-form-urlencoded
Connection: keep-alive
Accept: application/json
User-Agent: Chroma
Accept-Language: 
Content-Length: 20
Accept-Encoding: gzip, deflate
Request payload: {"COLOR":[0,255,95]}
<----- Request End -----
```
ESP8266-DEBUG;
```
POST / HTTP/1.1
Host: 192.168.1.171:8080
Content-Type: application/x-www-form-urlencoded
Connection: keep-alive
Accept: application/json
User-Agent: Chroma
Accept-Language: 
Content-Length: 20
Accept-Encoding: gzip, deflate
```

CODE WRITTEN ON ESP8266
```
def webServer():
    micropython_optimize=False

    s = socket.socket()
    ai = socket.getaddrinfo("192.168.1.171", 8080)
    print("Bind address info:", ai)
    addr = ai[0][-1]
    s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    s.bind(addr)
    s.listen(2)
    while True:
        res = s.accept()
        client_sock = res[0]
        client_addr = res[1]
        # print("Client address:", client_addr)
        # print("Client socket:", client_sock)

        req = client_sock.recv(1024)
        print("Request: %s" % req.decode())
        
        client_sock.send("HTTP/1.0 200 OK\r\n\r\n'Color: JSON'\r\n")

        client_sock.close()
        print()

```

dpgeorge, 2018-02-07:
Your `webServer()` function works OK for me on ESP8266 (and also on the unix port of MicroPython).  Running it on the ESP8266 (with adjusted IP address) gives the following output from the ESP8266:
```
Bind address info: [(2, 1, 0, '', ('192.168.0.2', 8080))]
Request: POST / HTTP/1.1
Host: 192.168.0.2:8080
User-Agent: curl/7.58.0
Accept: */*
Content-Length: 3
Content-Type: application/x-www-form-urlencoded

xyz

```
And I used curl to do the post with the command:
```
$ curl -d xyz 192.168.0.2:8080
'Color: JSON'
```

Sunrise17, 2018-02-07:
I think this is caused by "Accept-Encoding: gzip, deflate" encoding of Swift, micro python simple http server does not decode it.

dpgeorge, 2018-02-07:
> I think this is caused by "Accept-Encoding: gzip, deflate" encoding of Swift

That's unlikely.  "accept-encoding" is just saying what the client supports but the server can choose not to compress if it doesn't want to.

Please note that `client_sock.recv(1024)` may not read all the available data.  It may just return the first few bytes.  So you might need to make repeated calls to recv() to get everything.

Sunrise17, 2018-02-07:
Dear @dpgeorge , thanks for your response, i will check about it. I have one question regarding to asyncio method. Here is my code, and it works successfully. For ex. 1st request is coming then colorpicker() method is activated (its not loop method), and then when i send 2nd request, wipeReturn() method is activated (this is loop method). Here is my issue, then when i send 1st request again, wipeReturn() has not been stopped, i want him to be stopped or paused. How can i apply that?

```
async def handle_client(reader, writer):
    req = await reader.read()
    if "a" in req.decode():
        yield from colorPick("128")
        yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\n'Color: is applied'\r\n")
    elif "b" in req.decode():
        yield from wipeReturn(2)
        yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\n'Color: is applied'\r\n")

async def colorPick(color):
    print('colorPick')
    np.fill(color)
    np.write()

async def wipeReturn(delay):
    while True:
        print('wipereturn')
        await asyncio.sleep(delay)

loop = asyncio.get_event_loop()
loop.create_task(asyncio.start_server(handle_client, ipAddress, 8080))
loop.run_forever()
```

Sunrise17, 2018-02-14:
I have changed to ESP8266(512kb) to ESP32(with 4MB) now everything goes well, just sometimes, client is sending JSON but, ESP32 received as empty (after \r\n\r\n no message seen), its especially happening when the device is overloaded.

closed 2018-02-14 by Sunrise17